woodemi/quick_blue #29: Support for Linux?

flukejones, 2020-11-23:
Canonical has implemented a Linux Bluetooth support [package](https://github.com/canonical/bluez.dart) using bluez.

Sunbreak, 2020-11-23:
`QuickBlue` is mainly maintained for `NotepadCore` for ourselves.
We would consider supporting Web before Linux, as there is Web support in `NotepadCore`: https://github.com/woodemi/notepad_core/blob/cbf649b44576b808a5fd5d9ce0a2e592492e84a6/notepad_core_web/lib/notepad_core_web.dart#L34

Not quite familiar with Linux & `bluez`. Any contributions are welcome

I'll leave the issue open for further reference

Sunbreak, 2022-03-20:
FYI @rohitsangwan01 is working on https://github.com/woodemi/quick_blue/pull/63

Sunbreak, 2022-03-21:
fix by #61

closed 2022-03-21 by Sunbreak

CodeDoctorDE, 2022-05-22:
Hello, it would be cool if `setValue` can be supported by Linux.
I want to communicate cross platform between my apps. Thanks!

Sunbreak, 2022-05-23:
Maybe another issue?