woodemi/quick_blue #32: Steps to execute an example for windows

Balrog16, 2020-11-23:
May I please if there are any particular sequence of steps that I need to execute to run this code in windows 10?
At the moment, I am using vs code and have opened 
[main.dart](https://github.com/woodemi/quick_blue/blob/master/quick_blue_example/lib/main.dart) and tried compiling. I get the following error!
FileSystemException: Cannot create link, path = 'E:\appdev\expts\windows\quick_blue\quick_blue_example\windows\flutter\ephemeral\.plugin_symlinks\quick_blue_windows' (OS Error: Incorrect function.
, errno = 1)

Thank you for the time

Sunbreak, 2020-11-23:
You'll need Win10+ system and Flutter SDK on dev channel with `--enable-windows-desktop`

Then `flutter run` under directory `quick_blue_example` should work

More about [Desktop support for Flutter](https://flutter.dev/desktop)

Balrog16, 2020-11-23:
Thank you @Sunbreak, The issue seems to have it's origins in the fact that I am trying to use an external SSD for this purpose. Of course, I don't know the root cause though. I will try as you suggested

closed 2020-11-23 by Sunbreak