woodemi/quick_blue #33: Failed to send request:

husamhamu, 2020-11-27:
I ran into this problem after the first successful run for the "quick example" on windows 
`Failed to send request: {"jsonrpc":"2.0","id":"25","method":"ext.flutter.inspector.getSelectedSummaryWidget","params":{"objectGroup":"selection_5","isolateId":"isolates/751635408800707"}}`

I couldn't also connect and send data to any device

Sunbreak, 2020-11-27:
I seems that some cache cracks on Fluttter for Windows. When encountering similar problems, I try `git clean -xdf` then `flutter run` again

husamhamu, 2020-11-28:
Ok I will try that. But still can't send data to the device. as soo as I press the send button I lose the connection immediately

Sunbreak, 2020-11-28:
Check the `servicUUID`, `charateristicUUID` and the data. The code uses Woodemi's https://github.com/woodemi/notepad_core as an example.

It probably dont fit with your device

husamhamu, 2020-11-28:
I am using HC05 module

Sunbreak, 2020-11-28:
> I am using HC05 module

Sorry I've no idea what it is

husamhamu, 2020-12-01:
For those who might encounter the same problem I found the following link useful
[https://developer.android.com/reference/android/bluetooth/BluetoothDevice#createInsecureRfcommSocketToServiceRecord(java.util.UUID)](url)
I guess I can use the well-known SPP UUID 00001101-0000-1000-8000-00805F9B34FB with this Bluetooth model: HC-05

closed 2020-12-01 by Sunbreak