woodemi/quick_blue #34: Name of the device isn't correct

husamhamu, 2020-12-01:
For example instead of getting the name: HC-05 I got values like: 98 etc.

Sunbreak, 2020-12-01:
All platforms or only Windows?

husamhamu, 2020-12-01:
I guess only windows, because I didn't try other platforms

Sunbreak, 2020-12-02:
Could you try on Android/iOS if you have a mobile development environment?

Or try LightBlue: https://punchthrough.com/lightblue/ -- [iOS](https://apps.apple.com/us/app/lightblue/id557428110), [Android](https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer&hl=en_US)

husamhamu, 2020-12-02:
Ok , I tried LightBlue on Android. It is also not working for me. It can't find the device that I am trying to connect to

Sunbreak, 2020-12-02:
It is probably that the device set the name characteristic using wrong encoding.
You'd better debug the device with LigthBlue first.

husamhamu, 2020-12-02:
Actually I am using flutter_bluetooth_serial plugin on Android and it worked fine. That's why I think the problem is from quick_blue plugin.

Sunbreak, 2020-12-02:
[flutter_bluetooth_serial](https://pub.dev/packages/flutter_bluetooth_serial): Flutter basic implementation for Classical Bluetooth (only `RFCOMM` for now).
[quick_blue](https://pub.dev/packages/quick_blue): A cross-platform (Android/iOS/macOS/Windows) `BluetoothLE` plugin for Flutter 

RfComm is quite different from BLE: https://stackoverflow.com/questions/50503701/reading-data-from-a-bluetooth-device-low-energy-rfcomm-python3

husamhamu, 2020-12-02:
Sorry!,  I didn't know this.

closed 2020-12-02 by husamhamu

Sunbreak, 2020-12-02:
https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothRfcommChat

Sunbreak, 2020-12-02:
There is an example for Windows API: https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothRfcommChat

Contributions are welcome.