woodemi/quick_blue #35: Can't run on windows

Coimbra1984, 2020-12-10:
Hi,

I tried to test quick_blue (cool project, looks promising) on windows. Unfortunately I get this error message:
D:\temp\flutter\ble_tst>flutter run -d windows
Launching lib\main.dart on Windows in debug mode...
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: Der Befehl "setlocal [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -DBUILD_TYPE=Debug -P cmake_install.cmake [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: :cmEnd [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: :cmErrorLevel [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: exit /b %1 [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: :cmDone [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(153,5): error MSB3073: :VCEnd" wurde mit dem Code 1 beendet. [D:\temp\flutter\ble_tst\build\windows\INSTALL.vcxproj]
Building Windows application...
Exception: Build process failed.

Do you have any hints? This is my flutter doctor -v output:
D:\temp\flutter\ble_tst>flutter doctor -v
[√] Flutter (Channel dev, 1.25.0-8.0.pre, on Microsoft Windows [Version 10.0.18363.1256], locale de-DE)
    • Flutter version 1.25.0-8.0.pre at C:\flutter
    • Framework revision a12e2a473a (2 days ago), 2020-12-08 23:13:04 -0500
    • Engine revision df39e5c515
    • Dart version 2.12.0 (build 2.12.0-133.0.dev)

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at C:\Users\marku\AppData\Local\Android\Sdk
    • Platform android-29, build-tools 29.0.2
    • ANDROID_HOME = C:\Users\marku\AppData\Local\Android\Sdk
    • Java binary at: D:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b04)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files (x86)\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.8.2)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.8.30717.126
    • Windows 10 SDK version 10.0.18362.0

[√] Android Studio (version 3.6)
    • Android Studio at D:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b04)

[√] VS Code (version 1.51.1)
    • VS Code at C:\Users\marku\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.17.0

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.18363.1256]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 87.0.4280.88
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 87.0.664.55

• No issues found!

Sunbreak, 2020-12-10:
I always test with beta channel. Not sure why it fails on `1.25.0-8.0.pre`, which it is the latest dev today.

I'll try that later this weekend

Sunbreak, 2020-12-11:
I tried and it worked

```
# setup flutter alias
doskey flutter=C:\Users\sunbreak\flutter\flutter-1.25.x-dev\bin\flutter $*

git clone https://github.com/woodemi/quick_blue -b develop
cd quick_blue_example
flutter run
```

flutter doctor -v

```
C:\Users\sunbreak\workspace\woodemi\quick_blue-desktop\quick_blue_example>flutter doctor -v
[√] Flutter (Channel unknown, 1.25.0-8.0.pre, on Microsoft Windows [Version 10.0.18363.1198], locale en-US)
    • Flutter version 1.25.0-8.0.pre at C:\Users\sunbreak\flutter\flutter-1.25.x-dev
    • Framework revision a12e2a473a (2 days ago), 2020-12-08 23:13:04 -0500
    • Engine revision df39e5c515
    • Dart version 2.12.0 (build 2.12.0-133.0.dev)
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
    • Android SDK at C:\Users\sunbreak\AppData\Local\Android\Sdk
    • Platform android-29, build-tools 29.0.3
    • ANDROID_HOME = C:\Users\sunbreak\AppData\Local\Android\Sdk
    • ANDROID_SDK_ROOT = C:\Users\sunbreak\AppData\Local\Android\Sdk
    • Java binary at:
      C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\AndroidStudio\ch-0\201.6953283\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)
    • All Android licenses accepted.

[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.7.5)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.7.30523.141
    • Windows 10 SDK version 10.0.18362.0

[√] Android Studio (version 4.1.0)
    • Android Studio at C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\AndroidStudio\ch-0\201.6953283
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)

[√] IntelliJ IDEA Ultimate Edition (version 2020.3)
    • IntelliJ at C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\IDEA-U\ch-0\203.5981.155
    • Flutter plugin version 51.0.4
    • Dart plugin version 203.5981.152

[√] VS Code (version 1.51.1)
    • VS Code at C:\Users\sunbreak\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.16.0

[√] Connected device (1 available)
    • Windows (desktop) • windows • windows-x64 • Microsoft Windows [Version 10.0.18363.1198]

• No issues found!
```

Sunbreak, 2020-12-11:
@Coimbra1984 Could you run `flutter run -d windows -v`?

Coimbra1984, 2020-12-11:
okay, this is really strange. I deleted my app folder and created it again from scratch. Now it works... Sorry for the confusion.

closed 2020-12-11 by Coimbra1984

Sunbreak, 2020-12-11:
Did you open up the *.sln in build\windows of project root?
Visual Studio would mess up the cached build files in my case and cause similar error.
I prefer edit C++ within VS Code. VS Code's extension of CMake works like a charm

Coimbra1984, 2020-12-11:
No I only run the flutter command line tools
flutter create ble_test
flutter run -d windows -v
But now I remember, tthat I updated flutter in parallel, maybe this messed up the tools.

Sunbreak, 2020-12-11:
`flutter clean` always helps

Coimbra1984, 2020-12-11:
Cool, thx. I didn't know that.