woodemi/quick_blue #36: on Android, permission BLUETOOTH_PRIVILEGED ?

jdar, 2020-12-22:
Context: trying to run the example folder, on android 11

When I build and run the app, and click "startScan"
OR
When I add bluetooth permissions to AndroidManifest, build and run the app, and click "startScan"

Then the debug log shows a permission error (below). There are some very old comments that the BLUETOOTH_PRIVILEDGED permission is not available for third-party apps. Perhaps there is a workaround for using some of these functions?

D/BluetoothAdapter(13764): isLeEnabled(): BLE_ON
D/BluetoothLeScanner(13764): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
W/Binder  (13764): Caught a RuntimeException from the binder stub implementation.
W/Binder  (13764): java.lang.SecurityException: Need BLUETOOTH_PRIVILEGED permission: Neither user 10452 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
W/Binder  (13764):      at android.os.Parcel.createExceptionOrNull(Parcel.java:2373)
W/Binder  (13764):      at android.os.Parcel.createException(Parcel.java:2357)
W/Binder  (13764):      at android.os.Parcel.readException(Parcel.java:2340)
W/Binder  (13764):      at android.os.Parcel.readException(Parcel.java:2282)
W/Binder  (13764):      at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:1377)
W/Binder  (13764):      at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:457)
W/Binder  (13764):      at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:115)
W/Binder  (13764):      at android.os.Binder.execTransactInternal(Binder.java:1159)
W/Binder  (13764):      at android.os.Binder.execTransact(Binder.java:1123)

I have tried with and without AndroidManifest permissions: 

<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
<uses-feature android:name="android.hardware.bluetooth_le"
        android:required="false" />

Sunbreak, 2020-12-22:
You should turn on the bluetoot first. Or you'll get the ` java.lang.SecurityException: Need BLUETOOTH_PRIVILEGED permission`

https://stackoverflow.com/a/49502483/8017194

Sunbreak, 2020-12-23:
`Add isBluetoothAvailable to QuickBlue` https://github.com/woodemi/quick_blue/pull/37

jdar, 2020-12-23:
this works. Thanks!

closed 2020-12-23 by jdar