woodemi/quick_blue #39: OutputDebugString display

Balrog16, 2020-12-25:
How do I see these messages printed using "OutputDebugString".
I see that when I run a standalone visual stuido project, the console displays them. But in flutter project, is there a way to enable these prints? Thank you

Sunbreak, 2020-12-25:
# Solution 1

Open `%YOU_RPOJECT%.sln` via Visual Studio in `%PROJECT_ROOT%\build\windows` and you'll see several projects

- ALL_BUILD
- flutter_assemble
- flutter_wrapper_app
- flutter_wrapper_plugin
- INSTALL
- %YOU_RPOJECT%
- ZERO_CHECK

Right click on `%YOU_PROJECT%` project and debug, then you'll see the `OutputDebugString` in the output view

# Solution 2

https://docs.microsoft.com/en-us/sysinternals/downloads/debugview

https://support.sophos.com/support/s/article/KB-000034865?language=en_US

Balrog16, 2020-12-25:
Thank you so much!

closed 2020-12-25 by Balrog16