woodemi/quick_blue #41: Unable to use on Windows

Danau5tin, 2021-04-02:
```
NUGET.EXE not found.
CMake Error at flutter/ephemeral/.plugin_symlinks/quick_blue_windows/windows/CMakeLists.txt:10 (message):
  Please install this executable, and run CMake again.

Exception: Unable to generate build files

```

Sunbreak, 2021-04-02:
You should install https://www.nuget.org/ in you %PATH%

OR manully run `Install-Package Microsoft.Windows.CppWinRT` in [Visual Studio Package Manage Console](Package Manager Console) after you open the generated `runner.sln` in example's build directory

Sunbreak, 2021-08-06:
Check https://github.com/woodemi/quick_notify/issues/5#issuecomment-888743117 and https://github.com/woodemi/quick_notify/discussions/8

closed 2021-08-06 by Sunbreak