woodemi/quick_blue #42: JBL Go blutooth speaker not show in scan result

sauravedison, 2021-08-25:


Sunbreak, 2021-08-25:
Could your scan with LigthBlue and post the information of `JBL Go` device?

LightBlue manual: https://punchthrough.com/lightblue/

- Android: https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer
- iOS/macOS for US: https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer

sauravedison, 2021-08-25:
Hi,

Just check with LigthBlue Android app it is only display IOS device not display android and JBL GO Speaker.
 
JBL Go : https://www.amazon.in/JBL-Portable-Wireless-Bluetooth-Speaker/dp/B00TFGWAA8

Sunbreak, 2021-08-25:
I'm sorry if `quick_blue` is the same as LightBlue.

Some HID device is not discoverable via iOS's BLE APIs. The system handle those HID scan results itself

There was a hack way to get the result. I'm not sure whether it works still

Sunbreak, 2021-08-25:
https://colemancda.github.io/2018/03/25/Hacking-IOBluetooth

sauravedison, 2021-08-25:
I have run quick_blue in android device which works find but when i scan i got near by IOS device only (not get android device and other bluetooth device)

Why it is not finding near by android device as well ?

In Android Native Code working perfectly all this things.(which i have create demo)

Sunbreak, 2021-08-25:
Sorry I'm too busy these days to resolve the issue. PRs are welcome

Sunbreak, 2022-02-04:
I'd like to close it without further info

closed 2022-02-04 by Sunbreak