woodemi/quick_blue #44: Cannot run quick_blue_example app properly

BbsonLin, 2021-09-15:
On my none development Windows PC, quick_blue_example Release build cannnot run...

[Issue Video(none development PC)](https://youtu.be/x7umZ6ICow4)

--

And on my development PC, it sometimes got crashed when I running the `quick_blue_example.exe` file.

[Issue Video(development PC)](https://youtu.be/l2tRKOyNtMk)

also crashed during scanning devices sometimes.

Sunbreak, 2021-09-15:
> On my none development Windows PC, quick_blue_example Release build cannnot run...
> 
> [Issue Video(none development PC)](https://youtu.be/x7umZ6ICow4)

According to https://docs.microsoft.com/en-us/cpp/windows/redistributing-visual-cpp-files, you need to package all dependency yourself, e.g. [Microsoft Visual C++ Redistributable for Visual Studio 2015, 2017 and 2019](https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0) or MSVC Runtime

---

Useful link:
- https://knowledge.autodesk.com/support/autocad/learn-explore/caas/sfdcarticles/sfdcarticles/How-to-remove-and-reinstall-Microsoft-Visual-C-Runtime-Libraries.html
- https://docs.oracle.com/en/applications/jd-edwards/microsoft-windows/9.2/eoiwd/appendix-c---understanding-microsoft-visual-c-runtime-libraries.html#appendix-c---understanding-microsoft-visual-c-runtime-libraries

---

Package for Flutter Desktop bundling may help: https://github.com/anaisbetts/squirrel-flutter

jiangshan427, 2021-11-27:
_subscription = QuickBlue.scanResultStream.listen((result) { });
  QuickBlue.startScan();

As BbsonLin said the program sometimes  crashed  suddenly at the above two lines。Sometimes it works normally，so what's the problem?

Sunbreak, 2021-11-29:
Any crash log? Or you could use https://github.com/woodemi/quick_breakpad to collect some native crash

PS：加微信wk2311007聊更方便

jiangshan427, 2021-11-29:
感谢您的答复！我再去试试如何获取本地错误日志。学习下quick_breakpad 的使用。
控制台没有有效的日志输出。另外一个现象就是，发布Release版之后，崩溃概率降低。可能20次有1次。此外Release版启动之后 Bluetooth init 状态值是 false 。但是在IDEA里面运行的时候，这个值打开时就显示为true。Launching lib\main.dart on Windows in debug mode...
Building Windows application...
Debug service listening on ws://127.0.0.1:55532/NNzrh4TCh_A=/ws
Syncing files to device Windows...
Lost connection to device.

Sunbreak, 2021-11-29:
File another issue if you get any log

closed 2021-11-29 by Sunbreak

jiangshan427, 2021-12-10:
The following code should be added to the Functions  OnListenInternal and OnCancelInternal
 if (arguments)
  {
 xxxxx main code xxxxxxxx
}

Sunbreak, 2021-12-10:
I dont get the point. Could you pod the relative code?

PS：加微信wk2311007聊更方便

Sunbreak, 2021-12-21:
> The following code should be added to the Functions OnListenInternal and OnCancelInternal if (arguments) { xxxxx main code xxxxxxxx }

Done in https://github.com/woodemi/quick_blue/commit/f7349dddd00ee164f51a58f25750aeb1fdbf1b19