woodemi/quick_blue #45: Version problems

lgLindstrom, 2022-01-22:
I get this among other similar errors when trying to use your package.
Any plans to update ?

```
Because test <1.3.0 requires SDK version >=1.8.0 <2.0.0-∞ and test >=1.16.0-nullsafety <1.16.0-nullsafety.8 requires SDK version >=2.10.0-0 <2.12.0, test <1.3.0-∞ or >=1.16.0-nullsafety <1.16.0-nullsafety.8 is forbidden.
And because test >=1.16.0-nullsafety.8 <1.16.0-nullsafety.18 depends on typed_data >=1.3.0-nullsafety <1.3.0 and test >=1.16.0-nullsafety.18 <1.16.0-nullsafety.19 depends on test_api 0.2.19-nullsafety.7, test <1.3.0-∞ or >=1.16.0-nullsafety <1.16.0-nullsafety.19 requires typed_data >=1.3.0-nullsafety <1.3.0 or test_api 0.2.19-nullsafety.7.
And because test >=1.3.0 <1.12.0 depends on boolean_selector ^1.0.0 and test >=1.12.0 <1.13.0 depends on test_api 0.2.14, test <1.13.0-∞ or >=1.16.0-nullsafety <1.16.0-nullsafety.19 requires typed_data >=1.3.0-nullsafety <1.3.0 or test_api 0.2.14 or 0.2.19-nullsafety.7 or boolean_selector ^1.0.0.
And because test >=1.13.0 <1.14.5 depends on test_api 0.2.15 and test >=1.14.5 <1.15.0 depends on test_api 0.2.16, test <1.15.0-∞ or >=1.16.0-nullsafety <1.16.0-nullsafety.19 requires typed_data >=1.3.0-nullsafety <1.3.0 or test_api 0.2.14 or 0.2.15 or 0.2.16 or 0.2.19-nullsafety.7 or boolean_selector ^1.0.0.
And because test >=1.15.0 <1.15.3 depends on test_api 0.2.17 and test >=1.15.3 <1.15.5 depends on test_api 0.2.18, test <1.15.5-∞ or >=1.16.0-nullsafety <1.16.0-nullsafety.19 requires typed_data >=1.3.0-nullsafety <1.3.0 or test_api 0.2.14 or 0.2.15 or 0.2.16 or 0.2.17 or 0.2.18 or 0.2.19-nullsafety.7 or boolean_selector ^1.0.0.
And because test >=1.15.5 <1.16.0-nullsafety depends on test_api 0.2.18+1 and test >=1.16.0-nullsafety.19 <1.16.6 depends on test_api 0.2.19, test <1.16.6 requires test_api 0.2.14 or 0.2.15 or 0.2.16 or 0.2.17 or 0.2.18 or 0.2.18+1 or 0.2.19-nullsafety.7 or 0.2.19 or typed_data >=1.3.0-nullsafety <1.3.0 or boolean_selector ^1.0.0.
And because every version of flutter_test from sdk depends on both boolean_selector 2.1.0 and test_api 0.4.3, if flutter_test from sdk and test <1.16.6 then typed_data >=1.3.0-nullsafety <1.3.0.
And because every version of flutter from sdk depends on typed_data 1.3.0 and test >=1.16.6 depends on shelf_static ^1.0.0, if flutter_test from sdk and flutter from sdk and test any then shelf_static ^1.0.0.
And because shelf_static >=1.0.0 depends on convert ^3.0.0 and quick_blue >=0.2.0 depends on convert ^2.1.1, one of flutter_test from sdk or flutter from sdk or test any or quick_blue >=0.2.0 must be false.
And because psmc depends on both flutter from sdk and quick_blue ^0.2.0, flutter_test from sdk is incompatible with test.
So, because psmc depends on both test any and flutter_test from sdk, version solving failed.
```

Sunbreak, 2022-01-23:
Could you post your `flutter doctor -v`?

lgLindstrom, 2022-01-23:
flutter doctor -v
[√] Flutter (Channel stable, 2.8.1, on Microsoft Windows [Version 10.0.22000.434], locale sv-SE)
    • Flutter version 2.8.1 at C:\ProgramData\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 77d935af4d (5 weeks ago), 2021-12-16 08:37:33 -0800
    • Engine revision 890a5fca2e
    • Dart version 2.15.1

[√] Android toolchain - develop for Android devices (Android SDK version 30.0.2)
    • Android SDK at C:\Users\lglin\AppData\Local\Android\sdk
    • Platform android-30, build-tools 30.0.2
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)
    • All Android licenses accepted.

[X] Chrome - develop for the web (Cannot find Chrome executable at .\Google\Chrome\Application\chrome.exe)
    ! Cannot find Chrome. Try setting CHROME_EXECUTABLE to a Chrome executable.

[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.10.2)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.10.31410.357
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 4.1)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)

[√] VS Code (version 1.63.2)
    • VS Code at C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.32.0

[√] VS Code (version 1.56.0-insider)
    • VS Code at C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code Insiders
    • Flutter extension can be installed from:
       https://marketplace.visualstudio.com/items?itemName=Dart-Code.flutter

[√] Connected device (3 available)
    • BTV DL09 (mobile) • K6T6R16C01000368 • android-arm64  • Android 7.0 (API 24)
    • Windows (desktop) • windows          • windows-x64    • Microsoft Windows [Version 10.0.22000.434]
    • Edge (web)        • edge             • web-javascript • Microsoft Edge 97.0.1072.62

! Doctor found issues in 1 category.

lgLindstrom, 2022-01-23:
When I used "flutter_info_plus"  I hade this error 

[psmc] flutter pub get
Running "flutter pub get" in psmc...                            
Because package_info_plus >=1.0.4 depends on package_info_plus_platform_interface ^1.0.2 and no versions of package_info_plus_platform_interface match >1.0.2 <2.0.0, package_info_plus >=1.0.4 requires package_info_plus_platform_interface 1.0.2.
And because every version of version_widget from path depends on package_info_plus ^1.3.0 and package_info_plus_platform_interface 1.0.2 depends on plugin_platform_interface ^2.0.0, every version of version_widget from path requires plugin_platform_interface ^2.0.0.
And because quick_blue >=0.2.0 depends on quick_blue_platform_interface ^0.2.0 which depends on plugin_platform_interface ^1.0.1, version_widget from path is incompatible with quick_blue >=0.2.0.
So, because psmc depends on both quick_blue ^0.2.0 and version_widget from path, version solving failed.
pub get failed (1; So, because psmc depends on both quick_blue ^0.2.0 and version_widget from path, version solving failed.)
exit code 1

Sunbreak, 2022-01-23:
> When I used "package_info_plus" I hade this error
> 
> [psmc] flutter pub get Running "flutter pub get" in psmc... Because package_info_plus >=1.0.4 depends on package_info_plus_platform_interface ^1.0.2 and no versions of package_info_plus_platform_interface match >1.0.2 <2.0.0, package_info_plus >=1.0.4 requires package_info_plus_platform_interface 1.0.2. And because every version of version_widget from path depends on package_info_plus ^1.3.0 and package_info_plus_platform_interface 1.0.2 depends on plugin_platform_interface ^2.0.0, every version of version_widget from path requires plugin_platform_interface ^2.0.0. And because quick_blue >=0.2.0 depends on quick_blue_platform_interface ^0.2.0 which depends on plugin_platform_interface ^1.0.1, version_widget from path is incompatible with quick_blue >=0.2.0. So, because psmc depends on both quick_blue ^0.2.0 and version_widget from path, version solving failed. pub get failed (1; So, because psmc depends on both quick_blue ^0.2.0 and version_widget from path, version solving failed.) exit code 1

What about trying `dependency_overrides`: https://dart.dev/tools/pub/dependencies#dependency-overrides

lgLindstrom, 2022-01-23:
Yes, I could do that but the general advice about using not supported packages is => don't

I don't know if you are planning support this package or not?

It is the only ble package with desktop support so I hope you will continue and upgrade the package.

Sunbreak, 2022-01-23:
A little busy on Web support: https://github.com/woodemi/web_hid.dart

Could you PR on this problem?

lgLindstrom, 2022-01-23:
I never done a PR before and I don't have a clue about Dart packages. 

So I must repeat my question,, are you planning to support this package or drop it?

Sunbreak, 2022-01-23:
Going to upgrade but not recently

lgLindstrom, 2022-01-23:
Great, thanks.

codercq, 2022-02-09:
Windows Bluetooth plug is too few, the author update as soon as possible, you are the last hope of mankind

PR #46 (2022-02-28): Update to Null-Safety, Flutter >= 2.0
fix #40 , fix #45
closed 2022-02-28 by Sunbreak