woodemi/quick_blue #47: Exception when writing value on Windows

fotiDim, 2022-03-05:
I am facing an exception and crash each time I try to do `writeValue()`  to my device when running under Windows. Under Mac the same code works fine. The example app also crashes when I try to do `writeValue()`.
The Flutter console does not print anything useful upon crashing. Running the .sln in Visual Studio gives me this:

`Unhandled exception at 0x00007FF8BF42B382 (KernelBase.dll) in bt_remote.exe: 0xC000027B: An application-internal exception has occurred (parameters: 0x0000014B29305C30, 0x0000000000000002).`

[image]

I replicated the behavior on two different computers (Surface Pro + PC with BT Dongle).  Any idea what the problem might be?

Sunbreak, 2022-03-06:
Actually the coroutine support on C++17 is experimental feature of MSVC. It is

Sunbreak, 2022-03-06:
Or you could try catch the exception within `QuickBluePlugin::WriteValueAsync` like https://docs.microsoft.com/en-us/windows/uwp/cpp-and-winrt-apis/error-handling

```cpp
        try
        {
            auto thumbnail{ co_await imageFile.GetThumbnailAsync(FileProperties::ThumbnailMode::PicturesView) };
            if (thumbnail) bitmapImage.SetSource(thumbnail);
        }
        catch (winrt::hresult_error const& ex)
        {
            winrt::hresult hr = ex.code(); // HRESULT_FROM_WIN32(ERROR_FILE_NOT_FOUND).
            winrt::hstring message = ex.message(); // The system cannot find the file specified.
        }
```

fotiDim, 2022-03-06:
Thank you for the suggestions. I managed to get some more info by breaking when this exception type is thrown in VS. 

[image]

In my own code I am not doing anything fast paced. Just connecting with a button and then sending with another button.
As you can see in the video below the example app also crashes. Note that I have hardcoded my service and characteristic values.

[image]

I have noticed something weird though. If I have my device turned on, scan for it, turn it off and then connect, it still reports as connected. Also _handleServiceDiscovery never fires. I believe there is something going wrong with the connection process.

I also tried the [Bluetooth LE Explorer](https://github.com/microsoft/BluetoothLEExplorer) app for crosschecking and that works well with my device.

Sunbreak, 2022-03-06:
> I have noticed something weird though. If I have my device turned on, scan for it, turn it off and then connect, it still reports as connected. Also _handleServiceDiscovery never fires. I believe there is something going wrong with the connection process.

There is a little latecy within `BluetoothLEDevice_ConnectionStatusChanged`. That's all I know

fotiDim, 2022-03-06:
From what I see latency is not an issue. Connection Status events fire just fine. What I cannot understand is how it is possible to successfully connect to a device that is turned off. It seems that the connection is bogus and perhaps that is why writing values causes a crash.

Is there anything specific that should be taken into account when connecting?

Sunbreak, 2022-03-06:
That's a C++/WinRT problem. You'd better file issue on their repo or report on MSDN community forum

That's how I find solution my self. Actually there are few Docs about the Bluetooth APIs. Quite hard to search other than MSDN

fotiDim, 2022-03-06:
I will try to find a C++/WinRT BLE app and see if it works and how they do it.

I have one more finding regarding the connection. If I pair my device using Windows setting then it always appears as connected even though my device is Off. When I deleted my device from paired devices then it is reported correctly as connected or disconnected. The crash remains though. Are we supposed to pair our devices first before connection?

Since you have a working setup can please describe it? What kind of dongle and device are you using? I am trying to figure out if my windows version, dongle or peripheral are at fault.

Sunbreak, 2022-03-07:
> I will try to find a C++/WinRT BLE app and see if it works and how they do it.

Doc of samples: https://github.com/microsoft/windows-universal-samples/tree/main/Samples/BluetoothLE
Code of samples: https://github.com/microsoft/windows-universal-samples/tree/main/Samples/BluetoothLE

Some other help

- [Nordic Semiconductor](https://www.nordicsemi.com/): [Dev Zone](https://devzone.nordicsemi.com/), e.g. [this post](https://devzone.nordicsemi.com/f/nordic-q-a/48916/bluetooth-le-windows-10-using-winrt-c-code-works-if-device-not-paired-fails-with-unreachable-if-device-is-paired)
- LightBlue: [Windows tool](https://windowsden.uk/557428110/lightblue), [Docs](https://punchthrough.com/lightblue-features/)

> I have one more finding regarding the connection. If I pair my device using the Windows settings then it always appears as connected even though my device is Off. When I deleted my device from paired devices then it is reported correctly as connected or disconnected. The crash remains though. Are we supposed to pair our devices first before connection?

IIRC, there was a version restriction on `Connection without pairing` with `C++/WinRT` and Windows 10

- https://answers.microsoft.com/en-us/windows/forum/all/reconnecting-a-paired-bluetooth-device-without/90b42dd3-2998-4393-9d89-1624e120502f
- https://stackoverflow.com/questions/55765090/windows-bluetooth-le-require-pairing-before-connection
- https://social.msdn.microsoft.com/Forums/azure/en-US/66d6fb43-e41e-4751-99fe-170b8f63ad22/uwpcc-uwp-bluetooth-communication-without-pairing?forum=wpdevelop

> Since you have a working setup can please describe it? Which version of Windows, version of Visual Studio, kind of dongle and peripheral are you using?

Here it is, but I haven't test the setup lately

<details>
<summary>flutter doctor -v</summary>

```
[✓] Flutter (Channel stable, 2.10.1, on Microsoft Windows [Version 10.0.19044.1526], locale en-US)
    • Flutter version 2.10.1 at C:\Users\sunbreak\google\flutter-2.10.x
    • Upstream repository https://github.com/flutter/flutter
    • Framework revision db747aa133 (4 weeks ago), 2022-02-09 13:57:35 -0600
    • Engine revision ab46186b24
    • Dart version 2.16.1
    • DevTools version 2.9.2
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at C:\Users\sunbreak\AppData\Local\Android\sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\AndroidStudio\ch-0\203.7784292\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)
    • All Android licenses accepted.

[✓] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[✓] Visual Studio - develop for Windows (Visual Studio Community 2019 16.9.2)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.9.31112.23
    • Windows 10 SDK version 10.0.19041.0

[✓] Android Studio (version 2020.3)
    • Android Studio at C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\AndroidStudio\ch-0\203.7784292
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)

[✓] Android Studio (version 4.2)
    • Android Studio at C:\Users\sunbreak\AppData\Local\JetBrains\Toolbox\apps\AndroidStudio\ch-0\202.7351085
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.8+10-b944.6842174)

[✓] VS Code (version 1.64.2)
    • VS Code at C:\Users\sunbreak\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.34.0

[✓] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19044.1526]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 98.0.4758.102
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 98.0.1108.43

[✓] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!
```
</details>

fotiDim, 2022-03-08:
@Sunbreak thanks for the info. We are on the same major versions, so this part looks good.

I managed to debug the VS project. I take back what I said about the connection as it does seem to work. The issue I am facing is multipart:
 
The characteristic I should be writing data into is "ff01". However when reading the characteristics from the peripheral on Windows it is returned as "0000ff01-0000-1000-8000-00805f9b34fb". On Mac OS it just returned as "ff01". The was a first reason of a crash. I mitigated it by passing "0000ff01-0000-1000-8000-00805f9b34fb" as the characteristic which I am not even sure if it is right.

Then, the next crash happens in:

```c#
auto writeValueStatus = co_await gattCharacteristic.WriteValueAsync(from_bytevc(value), GattWriteOption::WriteWithResponse);
```

If I change it to `GattWriteOption::WriteWithoutResponse` then it does not crash anymore but it also does not work.

Sunbreak, 2022-03-08:
> The characteristic I should be writing data into is "ff01". However when reading the characteristics from the peripheral on Windows it is returned as "0000ff01-0000-1000-8000-00805f9b34fb". On Mac [...]

There is a HACK info: on iOS/macOS some FAMOUS service/characteristic would be shorten. Be careful when comparing the UUID. I haven't finished the refactor of `notepad_core` to `quick_blue` yet

- iOS: https://github.com/woodemi/notepad_core/blob/b0e329f3d6e02f14f8a0e5e48a6ddb48e026b658/notepad_core/ios/Classes/NotepadCorePlugin.m#L6-L36

```objc
        NSString *withGss = [NSString stringWithFormat:@"0000%@-%@", obj.UUID.uuidStr, GSS_SUFFIX];
        return [obj.UUID.uuidStr isEqualToString:service] || [withGss isEqualToString:service];
```

- macOS: https://github.com/woodemi/notepad_core/blob/b0e329f3d6e02f14f8a0e5e48a6ddb48e026b658/notepad_core_macos/macos/Classes/NotepadCorePlugin.swift#L5-L26

```swift
        let s = self.services?.first {
            $0.uuid.uuidStr == service || "0000\($0.uuid.uuidStr)-\(GSS_SUFFIX)" == service
        }
```

---

On the Dart side you'd better call like: https://github.com/woodemi/notepad_core/blob/b0e329f3d6e02f14f8a0e5e48a6ddb48e026b658/notepad_core/lib/woodemi/WoodemiClient.dart#L245-L256

```dart
const CODE__SERV_BATTERY = "180f";
const CODE__CHAR_BATTERY_LEVEL = "2a19";

const SERV__BATTERY = "0000$CODE__SERV_BATTERY-$GSS_SUFFIX";
const CHAR__BATTERY_LEVEL = "0000$CODE__CHAR_BATTERY_LEVEL-$GSS_SUFFIX";
```

---

FAMOUS service/characteristic list

- Official: https://www.bluetooth.com/specifications/specs/gatt-specification-supplement-5/
- gist: https://gist.github.com/sam016/4abe921b5a9ee27f67b3686910293026

Sunbreak, 2022-03-08:
> Then, the next crash happens in:
> ```cpp
> auto writeValueStatus = co_await gattCharacteristic.WriteValueAsync(from_bytevc(value), GattWriteOption::WriteWithResponse);
> ```
> 
> If I change it to `GattWriteOption::WriteWithoutResponse` then it does not crash anymore but it also does not work.

Check if the characteristic is with/withoutResponse before you try

Android would do it automatically but iOS/macOS/Windows needs manually check

Sunbreak, 2022-03-08:
https://www.bluetooth.com/blog/4-essential-tools-for-every-bluetooth-low-energy-developer/

1. Client Emulator Apps

- LightBlue app ([iOS](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110), [macOS](https://apps.apple.com/us/app/lightblue/id557428110))
- Nordic nRF Connect app ([iOS](https://itunes.apple.com/us/app/nrf-connect/id1054362403), [Android](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en), [Desktop](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF-Connect-for-desktop))

2. Bluetooth Sniffer

- High-end/commercial: [Ellisys sniffers](http://www.ellisys.com/products/btcompare.php), [Teledyne LeCroy](http://teledynelecroy.com/frontline/) sniffers (formerly Frontline), the [Spanalytics PANalyzr](https://www.spanalytics.com/panalyzr)
- Low-cost: [TI CC2540 USB dongle sniffer](http://www.ti.com/tool/CC2540EMK-USB), [Nordic nRF sniffer](https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE), [Ubertooth One](http://ubertooth.sourceforge.net/hardware/one/)

3. Power Measurement Tools

- Digital Multimeter (DMM): Useful for measuring average current draw and peak current (e.g. [Fluke 287 TRMS Multimeter](http://en-us.fluke.com/products/digital-multimeters/fluke-287-digital-multimeter.html))
- Oscilloscope: Used to get more accurate measurements and finer tuning of your power consumption

4. Bluetooth Specification Documents

Sunbreak, 2022-03-10:
> Then, the next crash happens in:
> 
> ```cs
> auto writeValueStatus = co_await gattCharacteristic.WriteValueAsync(from_bytevc(value), GattWriteOption::WriteWithResponse);
> ```
> 
> If I change it to `GattWriteOption::WriteWithoutResponse` then it does not crash anymore but it also does not work.

Thanks for your remind. I fix a mistake in https://github.com/woodemi/quick_blue/commit/a5bd82c81242677e743076cddc918205115d4c86

fotiDim, 2022-03-11:
I finally managed to make it work. I have made the following observations:
- [ ] Sending a wrong service or characteristic causes the library to crash.
- [ ] My device **requires** pairing for writing values. Without pairing trying to write a value causes the device to disconnect and the library to crash. Apparently this can be auto detected as[ Bluetooth LE Explorer](https://github.com/microsoft/BluetoothLEExplorer) shows

[image]

- [ ] I need a way to perform in-app pairing. There is the the [PairAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.pairasync?view=winrt-22000#windows-devices-enumeration-deviceinformationpairing-pairasync) method for in-app pairing.
- [ ] Alternatively to the previous one, I could monitor the pairing status so that I prompt the user to pair through Windows settings. There is the [isPaired](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.ispaired?view=winrt-22000#windows-devices-enumeration-deviceinformationpairing-ispaired) property.
- [ ] If I have my device turned on, scan for it, turn it off and then try to connect, the library mistakenly  reports that it connected successfully. [BluetoothLEExplorer](https://github.com/microsoft/BluetoothLEExplorer) handles that scenario correctly so it does not  seem to be an API limitation.

I can open separate issues for the above. The first one at least is straightforward to solve.

Sunbreak, 2022-03-12:
Thanks a lot for sharing!

Sunbreak, 2022-03-12:
> I can open separate issues for the above. The first one at least is straightforward to solve.

That would be nice

fotiDim, 2022-03-12:
Closing this as I opened more fine grained issues. There is a lot of useful information in this thread that you shared. This information could be added to the readme or a wiki page.

closed 2022-03-12 by fotiDim

Sunbreak, 2022-03-12:
> Closing this as I opened more fine grained issues. There is a lot of useful information in this thread that you shared. This information could be added to the readme or a wiki page.

I'm not sure. Could PR on the README so we could discuss there?