woodemi/quick_blue #53: [Windows] Getting a wrong service or characteristic causes the library to crash.

fotiDim, 2022-03-12:
Is I try to get a non-existent service or characteristic then the library crashes on the Windows side. The problem lies in those 2 functions. Basically it never inserts a service or characteristic in the corresponding array and it tries to return null.

```
IAsyncOperation<GattDeviceService> GetServiceAsync(std::string service) {
    if (gattServices.count(service) == 0) {
      auto serviceResult = co_await device.GetGattServicesAsync();
      if (serviceResult.Status() != GattCommunicationStatus::Success)
        co_return nullptr;

      for (auto s : serviceResult.Services())
        if (to_uuidstr(s.Uuid()) == service)
          gattServices.insert(std::make_pair(service, s));
    }
    co_return gattServices.at(service);
  }
```

```
  IAsyncOperation<GattCharacteristic> GetCharacteristicAsync(std::string service, std::string characteristic) {
    if (gattCharacteristics.count(characteristic) == 0) {
      auto gattService = co_await GetServiceAsync(service);

      auto characteristicResult = co_await gattService.GetCharacteristicsAsync();
      if (characteristicResult.Status() != GattCommunicationStatus::Success)
        co_return nullptr;

      for (auto c : characteristicResult.Characteristics())
        if (to_uuidstr(c.Uuid()) == characteristic)
          gattCharacteristics.insert(std::make_pair(characteristic, c));
    }
    co_return gattCharacteristics.at(characteristic);
  }
```

Sunbreak, 2022-03-12:
Is there any peripheral I could test on with?

fotiDim, 2022-03-15:
I don't think that the peripheral matters. Try to connect to any device and then get a service or characteristic with a non-existent random string. You can use the example app for that. It should replicate the crash. If not then let me know and I will investigate further.

Sunbreak, 2022-03-16:
If not `nullptr` what the expected result do you prefer?

fotiDim, 2022-03-18:
Well, the important thing is that the library should not crash and that on the Dart side the Future of `writeValue()` etc. should return an error.

In the Windows plugin `GetCharacteristicAsync()` can return `nullptr ` and then `WriteValueAsync()` (and the rest of the methods) should check the return value of `GetCharacteristicAsync` and if it is null then throw an error on the Dart side.

Sunbreak, 2022-03-18:
Thanks for your advise. I'd like to add a flutter driver test to ensure that case implemented on ALL platforms.

Leave the issue until then

rohitsangwan01, 2022-03-20:
i also found this issue on MacOs 
Things i tried

Scan 
and Connect to device

read characteristic -> App Crashed

now again after connecting 
first discover Services
then read characteristic , and it worked

now if i read a characteristic , unknown to device
App Crashed -> same error as of first case

quick_blue_macos/QuickBlueMacosPlugin.swift:241: Fatal error: Unexpectedly found nil while unwrapping an Optional value

[image]