woodemi/quick_blue #54: [Windows] Crash when writing value to encrypted characteristic

fotiDim, 2022-03-12:
My device requires pairing for writing values. Without pairing trying to write a value causes the device to disconnect and the library to crash. Apparently this can be auto detected as[ Bluetooth LE Explorer](https://github.com/microsoft/BluetoothLEExplorer) shows
[image]

Sunbreak, 2022-03-16:
Could you find the logic that detect the logic?

fotiDim, 2022-03-18:
I believe it's [this](https://github.com/microsoft/BluetoothLEExplorer/blob/579a918ed0d6181378111f9dfa27bbc8db902f05/BluetoothLEExplorer/BluetoothLEExplorer/ViewModels/CharacteristicPageViewModel.cs#L672-L676) part.

Vishavjeet-clarity, 2022-06-29:
I am also facing the same issue in windows, the app just crashes without any logs and also write method doesn't launch pairing intent in android like nordic NRF Connect app does.
Unable to pair in both windows and android.
My device also requires pairing for writing values