woodemi/quick_blue #55: [Windows] In-app pairing

fotiDim, 2022-03-12:
I need a way to perform in-app pairing. There is the the [PairAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.pairasync?view=winrt-22000#windows-devices-enumeration-deviceinformationpairing-pairasync) method for in-app pairing.

fotiDim, 2022-03-15:
For some of the other platforms it seems that an in-app pairing API is not available. For iOS for example this is handled [automatically by the system](https://stackoverflow.com/a/33112945).

Sunbreak, 2022-03-16:
What about Android?

fotiDim, 2022-03-17:
There is [createBond](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#createBond()) for Android.

Sunbreak, 2022-03-17:
Any device I could test on?

Sunbreak, 2022-03-17:
For example, here is a example from Bluetooth.org: https://github.com/WebBluetoothCG/ble-test-peripheral-android

fotiDim, 2022-03-18:
Any peripheral that supports pairing would do. Headphones, car hands free kits, Xbox controllers are some examples.