woodemi/quick_blue #56: [Windows] Expose isPaired and canPair

fotiDim, 2022-03-12:
I would like to monitor the pairing status so that I prompt the user to pair through Windows settings if needed. There are the [isPaired](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.ispaired?view=winrt-22000#windows-devices-enumeration-deviceinformationpairing-ispaired) and the [carPair](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.canpair?view=winrt-22000) properties. I would be great of those would be exposed on the Flutter side.

Sunbreak, 2022-03-12:
Is there similiar APIs on other platforms?

fotiDim, 2022-03-15:
As I [read ](https://stackoverflow.com/questions/47546690/ios-bluetooth-pairing-request-dialog-can-i-know-the-users-choice) for iOS at least the answer is no. It could be indirectly inferred by trying to write some value and see if it succeeds or not. I guess https://github.com/woodemi/quick_blue/issues/54 needs to be addressed first in that case.

Yongle-Fu, 2022-03-21:
BluetoothLEAdvertisementWatcher VS DeviceWatcher      

// auto aqsFilter = BluetoothDevice::GetDeviceSelectorFromPairingState(true) + L" AND System.Devices.Aep.IsConnected:=System.StructuredQueryType.Boolean#True";
      auto aqsFilter = BluetoothDevice::GetDeviceSelectorFromPairingState(true);
      auto requestedProperties = {
        L"System.Devices.Aep.IsConnected",
        L"System.Devices.Aep.DeviceAddress"
      };
      deviceWatcher = DeviceInformation::CreateWatcher(aqsFilter, requestedProperties, DeviceInformationKind::AssociationEndpoint);

Sunbreak, 2022-03-21:
Is there any chance that you could solve the #43 ?

yyshiftall, 2022-04-04:
As you may know, there isn't API for monitor a pairing status in iOS. 
In Android, there is a Broadcast intent [ACTION_BOND_STATE_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#ACTION_BOND_STATE_CHANGED)

And also, each OS can retreive bonded devices from system.

#### iOS
[retrievePeripheralsWithIdentifiers:](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1519127-retrieveperipheralswithidentifie) and [retrieveConnectedPeripheralsWithServices:
](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipheralswith)

[Best Practices for Interacting with a Remote Peripheral Device](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html)

#### Android
[getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices())

[Find Bluetooth devices  |  Android Developers](https://developer.android.com/guide/topics/connectivity/bluetooth/find-bluetooth-devices)

ymmtyuhei, 2022-04-04:
As you may know, there isn't API for monitor a pairing status in iOS. 
In Android, there is a Broadcast intent [ACTION_BOND_STATE_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#ACTION_BOND_STATE_CHANGED)

And also, each OS can retrieve bonded devices from system.

#### iOS
[retrievePeripheralsWithIdentifiers:](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1519127-retrieveperipheralswithidentifie) and [retrieveConnectedPeripheralsWithServices:
](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipheralswith)

[Best Practices for Interacting with a Remote Peripheral Device](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html)

#### Android
[getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices())

[Find Bluetooth devices  |  Android Developers](https://developer.android.com/guide/topics/connectivity/bluetooth/find-bluetooth-devices)

Sunbreak, 2022-04-06:
Thanks a lot for the info

I'll take a look at it this weekend