woodemi/quick_blue #62: MTU change

thewizard1983, 2022-03-18:
is there a way to change MTU value and priority to increase data rate (Bluetooth 5)?

Sunbreak, 2022-03-18:
Which platform do you infer?

Android has [requestConnectionPriority(int)](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int))

I haven't finished refactor from `notepad_core` to `quick_blue`

https://github.com/woodemi/notepad_core/blob/b0e329f3d6e02f14f8a0e5e48a6ddb48e026b658/notepad_core/android/src/main/kotlin/io/woodemi/notepad_core/NotepadCorePlugin.kt#L128-L136

```kt
            "requestConnectionPriority" -> {
                val bleConnectionPriority = call.argument<String>("bleConnectionPriority")!!
                connectGatt?.requestConnectionPriority(when (bleConnectionPriority) {
                    "high" -> BluetoothGatt.CONNECTION_PRIORITY_HIGH
                    "lowPower" -> BluetoothGatt.CONNECTION_PRIORITY_LOW_POWER
                    else -> BluetoothGatt.CONNECTION_PRIORITY_BALANCED
                })
                result.success(null)
            }
```

thewizard1983, 2022-03-18:
Android is my priority, iOs the second.
Is 2M Phy channel configurable too?

Thanks

Sunbreak, 2022-03-18:
> 2M Phy channel

Any doc or testable device?

thewizard1983, 2022-03-18:
2MPhy is for increasing bluetooth data rate in Bluetooth 5.
Here is how to do it in android: https://source.android.com/devices/bluetooth/ble_advertising

Sunbreak, 2022-03-18:
So we could use an Android 8+ device as **peripheral** to test `2MPhy`？

thewizard1983, 2022-05-30:
Hi,
sorry for my delay. You can configure an Arduino with BLE in order to work onto 2M phy and test the library.

Sunbreak, 2022-05-30:
> Hi, sorry for my delay. You can configure an Arduino with BLE in order to work onto 2M phy and test the library.

Sorry. I don't have an Arduino nor Raspberry Pi right now