woodemi/quick_blue #65: [MacOs] deviceId is missing on notification callback

eugeniobaglieri, 2022-03-19:
Using setNotifiable the code not work as expected.
In the log console appears this on every charateristic notification:

`
peripheral:didUpdateValueForForCharacteristic 2A6E {length = 2, bytes = 0x7d08} error: nil
flutter: _handleConnectorMessage {characteristicValue: {characteristic: 2a6e, value: [122, 8]}}

════════ Exception caught by services library ══════════════════════════════════
The following _TypeError was thrown during a platform message callback:
type 'Null' is not a subtype of type 'String'
`

And the handler setted in setValueHandler() it's never get called.
The exception is caused by line 75 of `method_channel_quick_blue.dart` file
The deviceId is required, but not received

closed 2022-03-19 by Sunbreak

Sunbreak, 2022-03-19:
Thanks for remind. Any further fix for iOS?

eugeniobaglieri, 2022-03-19:
Yes, just created PR #67

Sunbreak, 2022-03-19:
> Can you make it available into pub.dev? 

`quick_blue 0.3.1+3` is released

PR #66 (2022-03-19): Deviceid fix
Resolve #65