woodemi/quick_blue #75: Different Bluetooth address format per platform

h3x4d3c1m4l, 2022-03-22:
Hi! I noticed Bluetooth address is a int64 (as String) on Windows, which I seem to be able to parse correctly:

```
import 'package:convert/convert.dart';
[...]
ByteData btAddressData = ByteData(8)..setUint64(0, int.parse(device.deviceId));
String formatted = hex.encode(btAddressData.buffer.asUint8List(2, 6));
```

But on Android it is already formatted as a string (like `AA:BB:CC:11:22:33`). Would you accept a PR to equalize this difference? If so, what would be preferred? Perhaps it could be even better to have the Bluetooth address as Uint8List in the model.

Side note: I have not yet tested this on iOS/iPadOS/macOS as I currently don't have these devices available.

Sunbreak, 2022-03-23:
PRs are welcome. I could test them when available

Sunbreak, 2022-03-28:
There may be more challenges than we thought: https://github.com/afauch/BluetoothWatcher/issues/1