woodemi/quick_blue #76: serviceHandler never triggered on windows?

Fexot, 2022-03-23:
Hi,
First thank you for providing this BLE package which will be very usefull :).
Nevertheless, Is there a complete implementation of the serviceHandler under windows?
When using it as described in the example, it seems not to be triggered. (see code below). May I did something wrong? 

```
class CustomClient extends GenericClient(){
    CustomClient (){
        QuickBlue.setConnectionHandler(_handleConnectionChange);
        QuickBlue.setServiceHandler(_handleServiceDiscovery);
        QuickBlue.setValueHandler(_handleValueChange);
    }
    
     _handleConnectionChange(String deviceId, BlueConnectionState state){ //works well
        if(state == BlueConnectionState.connected){
            QuickBlue.discoverServices(deviceId);
        }
    }

    void _handleServiceDiscovery(String deviceId, String serviceId) { //never triggered
        print('_handleServiceDiscovery $deviceId, $serviceId');
    }

    void _handleValueChange(String deviceId, String characteristicId, Uint8List value) { //works well
        print('_handleValueChange $deviceId, $characteristicId, ${value.toList()}');
    }

}
```

Regards

Sunbreak, 2022-03-23:
Yep. Windows API is a little different. It is unnecessary to discover services before using them for now

```cpp
  } else if (method_name.compare("discoverServices") == 0) {
    // TODO
    result->Success(nullptr);
```

And it is difficult to add a fake implementation for now.

I'm planning to refactor the whole Windows implementation in the early of next month

F3xil, 2022-03-23:
Thank you for the answer,

Ok, it means that it's not possible to discover and use custom services and characteristics "on the fly"  for now without previously knowing the specific uuids. great job anyway! :)

 Have you planned to add this functionality during the refactorization?

Thank you for your answers.

Sunbreak, 2022-03-23:
> Have you planned to add this functionality during the refactorization?

Will do. But it takes time. Windows APIs are too cumbersome

F3xil, 2022-03-23:
Of course...

Thank you for your answers.

Regards!

Sunbreak, 2022-03-29:
> Ok, it means that it's not possible to discover and use custom services and characteristics "on the fly" for now without previously knowing the specific uuids. great job anyway! :)

Could you try https://github.com/timsneath/win32/pull/392? Does it help?

F3xil, 2022-04-05:
Sorry for my late answer,

No it won't help, but it's fine, I can use quick_blue as is for the moment, and wait for improvements

xDefcon, 2023-03-06:
@Sunbreak are there any news about this matter? Could you give us some guidelines in how to help with this issue? I may be able to fix this issue and open a PR.

iBog, 2023-03-25:
Same issue, tryed to skip calling QuickBlue.discoverServices for windows platform 
and use subscribe to hardcoded service and characteristic, doesn't help...