woodemi/quick_blue #79: quick_blue incompatible with connectivity_plus

ahoelzemann, 2022-03-26:
Hi, 

thanks for the plugin. Quick_blue seems to be incompatible with connectivity_plus. Maybe you can update the dependencies?

```
Because no versions of quick_blue match >0.4.1+1 <0.5.0 and quick_blue 0.4.1+1 depends on quick_blue_linux ^0.1.2, quick_blue ^0.4.1+1 requires quick_blue_linux ^0.1.2.
And because no versions of quick_blue_linux match >0.1.2 <0.2.0, quick_blue ^0.4.1+1 requires quick_blue_linux 0.1.2.
And because quick_blue_linux 0.1.2 depends on bluez ^0.7.9 and no versions of bluez match >0.7.9 <0.8.0, quick_blue ^0.4.1+1 requires bluez 0.7.9.
And because bluez 0.7.9 depends on dbus ^0.6.0 and nm >=0.5.0 depends on dbus ^0.7.0, quick_blue ^0.4.1+1 is incompatible with nm >=0.5.0.
And because connectivity_plus >=2.2.1 depends on connectivity_plus_linux ^1.3.0 which depends on nm ^0.5.0, quick_blue ^0.4.1+1 is incompatible with connectivity_plus >=2.2.1.
```

Sunbreak, 2022-03-26:
try `dependency_overrides`

```yaml
dependency_overrides:
  bluez: ^0.8.0
```

Sunbreak, 2022-03-26:
We would like to keep compatible with Flutter 2.2 until 7/2/2022

ahoelzemann, 2022-03-27:
thanks for the quick workaround.

closed 2022-03-27 by ahoelzemann