woodemi/quick_blue #81: How to get  characteristics by using serviceId

rohitsangwan01, 2022-03-30:
Hey , as title suggests can we have a method like , discover characteristics

Sunbreak, 2022-03-30:
Any usecase?

rohitsangwan01, 2022-03-30:
Yes i was implementing a feature where i have to list out all services and their characteristics
Something similar to nRF connect app
Not urgent feature for now, but just asking in case if its possible
As we are already discovering characterics , so just an api to get list of it, or maybe in discoverServices methods
Instead of getting services , we can get an object containing serviceId and characteristicList

Sunbreak, 2022-03-30:
Probably. But not on my roadmap

Busy doing pure Dart implementation on Windows

rohitsangwan01, 2022-03-31:
Okk great , let me know if i can be helpful

Sunbreak, 2022-03-31:
I'd like to review your implementation on this

rohitsangwan01, 2022-03-31:
Okk Will try

rohitsangwan01, 2022-03-31:
@Sunbreak  done , added a PR #82

Sunbreak, 2022-03-31:
Could split this feature request into another PR?

ext-danny, 2022-07-03:
Is this api be available already, because I can see this on the example.

rohitsangwan01, 2022-07-03:
@ext-danny  You can recieve Characteristics from the Service Handler , call discover services and it will return serviceId and Characteristics list

ext-danny, 2022-07-03:
@rohitsangwan01 I'm using the latest version i.e,   quick_blue: ^0.4.1+1
but the third parameter is missing. if I follow the example like you said I get the error 

**The argument type 'void Function(String, String, List<String>)' can't be assigned to the parameter type 'void Function(String, String)?'.**

checking on the plugin code I can't see the third parameter

**typedef OnServiceDiscovered = void Function(String deviceId, String serviceId);**

rohitsangwan01, 2022-07-03:
@ext-danny try version 5 , check on Pub dev , its in bera version currently