woodemi/quick_blue #88: LateInitializationError: Field '_instance@xxxxxxxx' has not been initialized.

robnorr, 2022-04-10:
Hi,

I have a class which performs some QuickBlue operations, (e.g. QuickBlue.scanResultStream.listen(), QuickBlue.setConnectionHandler() etc.), however when these commands are reached, I get a LateInitializationError on _instance :

```
LateInitializationError: Field '_instance@35118451' has not been initialized.

When the exception was thrown, this was the stack:
#0      QuickBluePlatform._instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart)
#1      QuickBluePlatform.instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart:29:44)
#2      QuickBlue.scanResultStream (package:quick_blue/quick_blue.dart:20:30)
```

If I put the same code in a StatefulWidget rather than a class, there is no error. Does QuickBlue have to operate inside a stateful widget or is this an error?

Sunbreak, 2022-04-11:
Could try `WidgetsFlutterBinding.ensureInitialized();` first then call `QuickBlue` methods afterwards?

robnorr, 2022-04-11:
Unfortunately that doesn't seem to have changed anything. I've tried to create a minimal example of the issue here:

```
import 'dart:async'; // for StreamSubscription
import 'dart:typed_data'; // for Uint8List

import 'package:flutter/material.dart';

import 'package:quick_blue/quick_blue.dart';

void main() {
  runApp(const MyApp());
}

class MyApp extends StatelessWidget {
  const MyApp({Key? key}) : super(key: key);

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      title: 'Flutter Demo',
      theme: ThemeData(
        primarySwatch: Colors.blue,
      ),
      home: const MyHomePage(title: 'Flutter Demo Home Page'),
    );
  }
}

class MyHomePage extends StatefulWidget {
  const MyHomePage({Key? key, required this.title}) : super(key: key);

  final String title;

  @override
  State<MyHomePage> createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  BluetoothConnection bluetoothConnector = BluetoothConnection();

  void _buttonClick() {
    setState(() {
      bluetoothConnector.connect();
    });
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      appBar: AppBar(
        title: Text(widget.title),
      ),
      body: Center(
        child: Column(
          mainAxisAlignment: MainAxisAlignment.center,
        ),
      ),
      floatingActionButton: FloatingActionButton(
        onPressed: _buttonClick,
        tooltip: 'Connect',
        child: const Icon(Icons.add),
      ),
    );
  }
}

class BluetoothConnection
{
  BluetoothConnection()
  {
    _subscription = QuickBlue.scanResultStream.listen((result) {
      if (result.name == "BluetoothDevice") {
        device = result;
      }
    });
    QuickBlue.setConnectionHandler(_handleConnectionChange);
    QuickBlue.setServiceHandler(_handleServiceDiscovery);
    QuickBlue.setValueHandler(_handleValueChange);
  }

  StreamSubscription<BlueScanResult>? _subscription;
  BlueScanResult? device;

  void connect()
  {
    QuickBlue.startScan();
    while(device == null);
    QuickBlue.connect(device!.deviceId);
  }

  void disconnect() {
    QuickBlue.disconnect(device!.deviceId);
    _subscription?.cancel();
    QuickBlue.setValueHandler(null);
    QuickBlue.setServiceHandler(null);
    QuickBlue.setConnectionHandler(null);
  }

  void _handleConnectionChange(String deviceId, BlueConnectionState state) {
    print('_handleConnectionChange $deviceId, $state');
  }

  void _handleServiceDiscovery(String deviceId, String serviceId) {
    print('_handleServiceDiscovery $deviceId, $serviceId');
  }

  void _handleValueChange(String deviceId, String characteristicId, Uint8List value) {
    print('_handleValueChange $deviceId, $characteristicId, ${String.fromCharCodes(value)}');
  }
}
```

When I run this, I get the LateInitializationError immediately. Adding `WidgetsFlutterBinding.ensureInitialized();` to the constructor of `BluetoothConnection` doesn't change anything for me.

robnorr, 2022-04-11:
Alright, I found the issue. Unfortunately my dart/flutter understanding isn't good enough to know what/why/how this is set, or why it causes an issue but I'll post the fix here for anyone else:

In `.dart_tool/package_config_subset` change
```
file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.2/
file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.2/lib/
```
to
```
file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.1+1/
file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.1+1/lib/
```

And in `.dart_tool/package_config` change
```
      "rootUri": "file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.2",
```
to
```
      "rootUri": "file:///Users/user/.pub-cache/hosted/pub.dartlang.org/quick_blue_platform_interface-0.3.1+1",
```

btw, I forgot to mention in the original post I was building macos when I encountered this error.

closed 2022-04-11 by robnorr