woodemi/quick_blue #92: run time error in windows

krishnadevy57, 2022-04-20:
Hello Guys, 

 I am getting following issue in windows when trying to run app.

NUGET.EXE not found.
CMake Error at flutter/ephemeral/.plugin_symlinks/quick_blue_windows/windows/CMakeLists.txt:10 (message):
  Please install this executable, and run CMake again.

Exception: Unable to generate build files

please help me

Sunbreak, 2022-04-20:
https://github.com/woodemi/quick_notify/discussions/8

krishnadevy57, 2022-04-21:
Thanks Sunbreak for your reply , now getting following issue   set_target_properties Can not find target to add properties to:
  quick_blue_windows_plugin

CMake Error at flutter/ephemeral/.plugin_symlinks/quick_blue_windows/windows/CMakeLists.txt:12 (target_link_libraries):
  Cannot specify link libraries for target "quick_blue_windows_plugin" which
  is not built by this project.

Exception: Unable to generate build files

please help

Sunbreak, 2022-04-21:
Could you post `flutter doctor -v`?

And `flutter build windows -v` under `quick_blue_example`?

krishnadevy57, 2022-04-21:
D:\miridiam1> C:\Users\DELL>flutter build windows -v
>> [  +96 ms] executing: [C:\src\flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
>> [ +665 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
>> [   +3 ms] 13a2fb10b838971ce211230f8ffdd094c14af02c
>> [   +1 ms] executing: [C:\src\flutter/] git tag --points-at 13a2fb10b838971ce211230f8ffdd094c14af02c
>> [  +87 ms] Exit code 0 from: git tag --points-at 13a2fb10b838971ce211230f8ffdd094c14af02c
>> [   +1 ms] 2.13.0-0.1.pre
>> [  +17 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref --symbolic @{u}
>> [  +73 ms] Exit code 0 from: git rev-parse --abbrev-ref --symbolic @{u}
>> [   +1 ms] origin/dev
>> [   +2 ms] executing: [C:\src\flutter/] git ls-remote --get-url origin
>> [  +64 ms] Exit code 0 from: git ls-remote --get-url origin
>> [   +2 ms] https://github.com/flutter/flutter.git
>> [  +87 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref HEAD
>> [  +68 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
>> [   +1 ms] dev
>> [ +103 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
>> [   +6 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
>> [  +75 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'GradleWrapper' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterSdk' is not required, skipping update.
>> [   +5 ms] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
>> [   +2 ms] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
>> [   +1 ms] Artifact Instance of 'PubDependencies' is not required, skipping update.
>> [  +20 ms] "flutter windows" took 178ms.
>> [  +30 ms] Error: No pubspec.yaml file found.
>>            This command should be run from the root of your Flutter project.
>> [   +7 ms]
>>            #0      FlutterCommand.validateCommand (package:flutter_tools/src/runner/flutter_command.dart:1465:11)
>>            #1      FlutterCommand.verifyThenRunCommand (package:flutter_tools/src/runner/flutter_command.dart:1298:11)
>>            <asynchronous suspension>
>>            #2      FlutterCommand.run.<anonymous closure>
>> (package:flutter_tools/src/runner/flutter_command.dart:1183:27)
>>            <asynchronous suspension>
>>            #3      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
>>            <asynchronous suspension>
>>            #4      CommandRunner.runCommand (package:args/command_runner.dart:209:13)
>>            <asynchronous suspension>
>>            #5      FlutterCommandRunner.runCommand.<anonymous closure>
>>            (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
>>            <asynchronous suspension>
>>            #6      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
>>            <asynchronous suspension>
>>            #7      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
[... cut ...]
           #9      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
           <asynchronous suspension>
           #10     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_tools/executable.dart:94:3)
           <asynchronous suspension>

Sunbreak, 2022-04-21:
> [ +36 ms] executing: C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe -S
> D:\miridiam1\windows -B build\windows -G Visual [...]

- `cmake.exe -S D:\miridiam1\windows -B build\windows -G Visual Studio 17 2022`
- `set_target_properties Can not find target to add properties to: quick_blue_windows_plugin`
- `Cannot specify link libraries for target "quick_blue_windows_plugin"`

It seems that target `quick_blue_windows_plugin` is not recognized. Could try with Visual Studio 2019?

I haven't try VS 2022 and am not sure about the compatiblity

Sunbreak, 2022-04-21:
https://github.com/woodemi/quick_blue/runs/5860012305

Recheck the most recent GitHub Action: https://github.com/woodemi/quick_blue/runs/5860012305?check_suite_focus=true

```
Current runner version: '[2](https://github.com/woodemi/quick_blue/runs/5860012305?check_suite_focus=true#step:1:2).289.2'
Operating System
  Microsoft Windows Server 2022
  10.0.20[3](https://github.com/woodemi/quick_blue/runs/5860012305?check_suite_focus=true#step:1:3)[4](https://github.com/woodemi/quick_blue/runs/5860012305?check_suite_focus=true#step:1:4)[8](https://github.com/woodemi/quick_blue/runs/5860012305?check_suite_focus=true#step:1:8)
  Datacenter
Virtual Environment
  Environment: windows-2022
  Version: 20220330.1
  Included Software: https://github.com/actions/virtual-environments/blob/win22/20220330.1/images/win/Windows2022-Readme.md
  Image Release: https://github.com/actions/virtual-environments/releases/tag/win22%2F20220330.1
```

It should work

krishnadevy57, 2022-04-21:
ok let me try thanks

> * `cmake.exe -S D:\miridiam1\windows -B build\windows -G Visual Studio 17 2022`
> * `set_target_properties Can not find target to add properties to: quick_blue_windows_plugin`
> * `Cannot specify link [...]

thanks I will try

Sunbreak, 2022-04-21:
FYI: https://stackoverflow.com/questions/70212613/flutter-for-desktop-visual-studio-2022-problem

krishnadevy57, 2022-04-21:
[√] Flutter (Channel dev, 2.13.0-0.1.pre, on Microsoft Windows [Version 10.0.19044.1466], locale en-IN)
    • Flutter version 2.13.0-0.1.pre at C:\src\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 13a2fb10b8 (9 days ago), 2022-04-12 15:34:25 -0500
    • Engine revision 499984f99c
    • Dart version 2.17.0 (build 2.17.0-266.1.beta)
    • DevTools version 2.12.1

[√] Android toolchain - develop for Android devices (Android SDK version 32.1.0-rc1)
    • Android SDK at C:\Users\DELL\AppData\Local\Android\sdk
    • Platform android-32, build-tools 32.1.0-rc1
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Professional 2019 16.11.13)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional
    • Visual Studio Professional 2019 version 16.11.32413.511
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 2021.1)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)

[√] VS Code (version 1.66.2)
    • VS Code at C:\Users\DELL\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.38.1

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19044.1466]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 100.0.4896.127
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 100.0.1185.44

[√] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!
still same problem

krishnadevy57, 2022-04-21:
flutter build windows -v
[ +107 ms] executing: [C:\src\flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[ +690 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +1 ms] 13a2fb10b838971ce211230f8ffdd094c14af02c
[   +1 ms] executing: [C:\src\flutter/] git tag --points-at 13a2fb10b838971ce211230f8ffdd094c14af02c
[ +108 ms] Exit code 0 from: git tag --points-at 13a2fb10b838971ce211230f8ffdd094c14af02c
[        ] 2.13.0-0.1.pre
[  +21 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[  +78 ms] Exit code 0 from: git rev-parse --abbrev-ref --symbolic @{u}
[        ] origin/dev
[        ] executing: [C:\src\flutter/] git ls-remote --get-url origin
[  +71 ms] Exit code 0 from: git ls-remote --get-url origin
[        ] https://github.com/flutter/flutter.git
[  +94 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref HEAD
[  +71 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] dev
[ +104 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[  +10 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +91 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[        ] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +2 ms] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[   +7 ms] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[   +2 ms] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[   +2 ms] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'PubDependencies' is not required, skipping update.
[  +81 ms] Skipping pub get: version match.
[  +68 ms] Found plugin quick_blue at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue-0.4.1+1\
[   +8 ms] Found plugin quick_blue_linux at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_linux-0.1.2\
[   +5 ms] Found plugin quick_blue_macos at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_macos-0.3.1+2\
[   +5 ms] Found plugin quick_blue_windows at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_windows-0.3.2\
[ +111 ms] Found plugin quick_blue at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue-0.4.1+1\
[   +3 ms] Found plugin quick_blue_linux at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_linux-0.1.2\
[   +3 ms] Found plugin quick_blue_macos at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_macos-0.3.1+2\
[   +3 ms] Found plugin quick_blue_windows at C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\quick_blue_windows-0.3.2\
[ +113 ms] Generating D:\miridiam1\android\app\src\main\java\io\flutter\plugins\GeneratedPluginRegistrant.java
[ +115 ms]  Building with sound null safety 
[  +27 ms] executing: C:\Program Files (x86)\Microsoft Visual Studio\Installer\vswhere.exe -format json -products * -utf8 -latest -version 16
-requires Microsoft.VisualStudio.Workload.NativeDesktop Microsoft.VisualStudio.Component.VC.Tools.x86.x64
Microsoft.VisualStudio.Component.VC.CMake.Project
[... cut ...]
           #9      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
           <asynchronous suspension>
           #10     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_tools/executable.dart:94:3)
           <asynchronous suspension>

[ +227 ms] ensureAnalyticsSent: 213ms

Sunbreak, 2022-04-21:
What about `flutter-2.8.1` or `flutter-2.10.6`?

krishnadevy57, 2022-04-21:
[+14382 ms] Exception: Unable to generate build files
[  +10 ms] "flutter run" took 16,103ms.
[  +14 ms]
           #0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
           #1      RunCommand.runCommand (package:flutter_tools/src/commands/run.dart:699:9)
           <asynchronous suspension>
           #2      FlutterCommand.run.<anonymous closure> (package:flutter_tools/src/runner/flutter_command.dart:1183:27)
           <asynchronous suspension>
           #3      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #4      CommandRunner.runCommand (package:args/command_runner.dart:209:13)
           <asynchronous suspension>
           #5      FlutterCommandRunner.runCommand.<anonymous closure> (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
           <asynchronous suspension>
           #6      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #7      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
           <asynchronous suspension>
           #8      run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #9      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #10     main (package:flutter_tools/executable.dart:94:3)
           <asynchronous suspension>

after visual studio 2019

Sunbreak, 2022-04-21:
Could you try Flutter 2.8.1 or Flutter 2.10.5?

Or could you reboot your PC?

krishnadevy57, 2022-04-22:
yes tried but getting issues on windows

Sunbreak, 2022-04-22:
```
[  +25 ms] Generating D:\miridiam1\android\app\src\main\java\io\flutter\plugins\GeneratedPluginRegistrant.java
...
[  +26 ms] executing: C:\Program Files (x86)\Microsoft Visual
Studio\2019\Professional\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe -S D:\miridiam1\windows -B
build\windows -G Visual Studio 16 2019
```

What is `D:\miridiam1\`?

There **SHOULD** be `quick_blue_example` according to: https://github.com/woodemi/quick_blue/runs/5860012305

```
[  +10 ms] executing: C:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe -S D:\a\quick_blue\quick_blue\quick_blue_example\windows -B build\windows -G Visual Studio 17 2022
```

---

Could you try `quick_blue_example` exactly like below?

```
> git clone https://github.com/woodemi/quick_blue
> cd quick_blue_example
> flutter build windows -v
```

krishnadevy57, 2022-04-22:
D:\quick_blue-master\quick_blue-master\quick_blue_example>flutter build windows -v
Building flutter tool...
Running pub upgrade...
[ +139 ms] executing: [C:\src\flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[ +857 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[   +2 ms] 5464c5bac742001448fe4fc0597be939379f88ea
[   +3 ms] executing: [C:\src\flutter/] git tag --points-at 5464c5bac742001448fe4fc0597be939379f88ea
[ +560 ms] Exit code 0 from: git tag --points-at 5464c5bac742001448fe4fc0597be939379f88ea
[   +2 ms] 2.10.5
[  +37 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref --symbolic @{u}
[ +143 ms] Exit code 0 from: git rev-parse --abbrev-ref --symbolic @{u}
[   +2 ms] origin/stable
[   +1 ms] executing: [C:\src\flutter/] git ls-remote --get-url origin
[ +142 ms] Exit code 0 from: git ls-remote --get-url origin
[   +2 ms] https://github.com/flutter/flutter.git
[ +339 ms] executing: [C:\src\flutter/] git rev-parse --abbrev-ref HEAD
[  +45 ms] Exit code 0 from: git rev-parse --abbrev-ref HEAD
[        ] stable
[ +127 ms] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[   +5 ms] Artifact Instance of 'WindowsEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[  +73 ms] Artifact Instance of 'MaterialFonts' is not required, skipping update.
[        ] Artifact Instance of 'GradleWrapper' is not required, skipping update.
[        ] Artifact Instance of 'AndroidGenSnapshotArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'AndroidInternalBuildArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterWebSdk' is not required, skipping update.
[        ] Artifact Instance of 'FlutterSdk' is not required, skipping update.
[   +3 ms] Artifact Instance of 'WindowsUwpEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxEngineArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'LinuxFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'MacOSFuchsiaSDKArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'FlutterRunnerSDKArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FlutterRunnerDebugSymbols' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[        ] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[   +6 ms] Artifact Instance of 'IosUsbArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'FontSubsetArtifacts' is not required, skipping update.
[   +1 ms] Artifact Instance of 'PubDependencies' is not required, skipping update.
[ +134 ms] Skipping pub get: version match.
[  +90 ms] Found plugin quick_blue at D:\quick_blue-master\quick_blue-master\quick_blue\
[  +12 ms] Found plugin quick_blue_linux at D:\quick_blue-master\quick_blue-master\quick_blue_linux\
[   +5 ms] Found plugin quick_blue_macos at D:\quick_blue-master\quick_blue-master\quick_blue_macos\
[   +8 ms] Found plugin quick_blue_windows at D:\quick_blue-master\quick_blue-master\quick_blue_windows\
[ +107 ms] Found plugin quick_blue at D:\quick_blue-master\quick_blue-master\quick_blue\
[   +6 ms] Found plugin quick_blue_linux at D:\quick_blue-master\quick_blue-master\quick_blue_linux\
[   +2 ms] Found plugin quick_blue_macos at D:\quick_blue-master\quick_blue-master\quick_blue_macos\
[   +3 ms] Found plugin quick_blue_windows at D:\quick_blue-master\quick_blue-master\quick_blue_windows\
[  +13 ms] Generating D:\quick_blue-master\quick_blue-master\quick_blue_example\android\app\src\main\java\io\flutter\plugins\GeneratedPluginRegistrant.java
[ +117 ms]  Building with sound null safety
[  +25 ms] executing: C:\Program Files (x86)\Microsoft Visual Studio\Installer\vswhere.exe -format json -products * -utf8 -latest -version 16 -requires
Microsoft.VisualStudio.Workload.NativeDesktop Microsoft.VisualStudio.Component.VC.Tools.x86.x64 Microsoft.VisualStudio.Component.VC.CMake.Project
[... cut ...]
           #10     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #12     main (package:flutter_tools/executable.dart:94:3)
           <asynchronous suspension>

[ +265 ms] ensureAnalyticsSent: 258ms
[   +9 ms] Running shutdown hooks
[   +3 ms] Shutdown hooks complete
[   +6 ms] exiting with code 1

same with quick_blue_example

Sunbreak, 2022-04-22:
Some progress here. No more `quick_blue_windows_plugin` compliants

- Have you enabled `Developer mode` on your Windows 10? https://github.com/flutter/flutter/issues/73207
- Could you try https://github.com/flutter/flutter/issues/85143#issuecomment-867931944
- Or https://stackoverflow.com/questions/67932485/exception-unable-to-generate-build-files-for-window-build ?

Sunbreak, 2022-04-22:
BTW, I would recommand `Command Prompt` than `Power Shell` for Flutter development