woodemi/quick_blue #96: On Writing command getting crash

krishnadevy57, 2022-04-26:
Hello, 
 
I am getting error on writing command , please help how can I write a command like change name 

Thanks

Sunbreak, 2022-04-26:
It didn't find the right characteristic. Make sure you got the right UUID

https://github.com/woodemi/quick_blue/blob/f451df7ce21eb77074c6fdcbe2999ae7a933bbb7/quick_blue_macos/macos/Classes/QuickBlueMacosPlugin.swift#L25-L33

Could you post the service discovery log?

krishnadevy57, 2022-04-26:
centralManager:didDiscoverPeripheral nil 9B0C3D3F-3FDD-13C1-C217-2324BF12785B
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
centralManager:didDiscoverPeripheral nil 1CC44F1E-D218-DFC4-E8FE-81EA760E220D
peripheral: 719201F7-C5D4-1BCD-CC0D-8A36F131DA94 didDiscoverServices: nil
peripheral:didDiscoverCharacteristicsForService (180a, 2a24
peripheral:didDiscoverCharacteristicsForService (180a, 2a25
peripheral:didDiscoverCharacteristicsForService (180a, 2a27
peripheral:didDiscoverCharacteristicsForService (180a, 2a26
peripheral:didDiscoverCharacteristicsForService (180a, 2a28
peripheral:didDiscoverCharacteristicsForService (180f, 2a19
flutter: _handleServiceDiscovery 719201F7-C5D4-1BCD-CC0D-8A36F131DA94, 180a
peripheral:didDiscoverCharacteristicsForService (e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208, e5f50001-c0ec-46ba-bd1f-b4c9f0da4208
peripheral:didDiscoverCharacteristicsForService (e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208, e5f50003-c0ec-46ba-bd1f-b4c9f0da4208
peripheral:didDiscoverCharacteristicsForService (e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208, e5f50004-c0ec-46ba-bd1f-b4c9f0da4208
peripheral:didDiscoverCharacteristicsForService (fe59, 8ec90003-f315-4f60-9fb8-838830daea50
flutter: _handleServiceDiscovery 719201F7-C5D4-1BCD-CC0D-8A36F131DA94, 180f
flutter: _handleServiceDiscovery 719201F7-C5D4-1BCD-CC0D-8A36F131DA94, e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208
flutter: _handleServiceDiscovery 719201F7-C5D4-1BCD-CC0D-8A36F131DA94, fe59

krishnadevy57, 2022-04-26:
void updateFriendlyName()  async{
    var valuesKey = utf8.encode("F") ;
    var values = utf8.encode(friendlyNameTextController.value.text) ;

    await  QuickBlue.writeValue(connectedDeviceId, "e5f5f000", "e5f50001",
        Uint8List.fromList(valuesKey + values), BleOutputProperty.withResponse);

  }

Sunbreak, 2022-04-26:
> peripheral:didDiscoverCharacteristicsForService (e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208, e5f50001-c0ec-46ba-bd1f-b4c9f0da4208

Sunbreak, 2022-04-26:
e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208, e5f50001-c0ec-46ba-bd1f-b4c9f0da4208

krishnadevy57, 2022-04-26:
Thank you very much , issue  fixed