woodemi/quick_blue #98: Check if device is already connected 📲

JW94, 2022-04-30:
Hi, 

Thank you very much for your effort. Is there a way to see all connected devices? If a device is already connected, It won't be shown in the `scanResultStream` Stream.

Thanks in advance!

Sunbreak, 2022-05-02:
Which platform are you concerned?

JW94, 2022-05-02:
Hi Sunbreak,

I'm working on an iOS-App.

Thank you

HarryWindsor, 2022-05-04:
Hey @JW94 and @Sunbreak,

I have the same problem. I can not find devices that I was connected to before.
I have only tested iOS, as I do not own an Android device.

Sunbreak, 2022-05-06:
> Hey @JW94 and @Sunbreak,
> 
> I have the same problem. I can not find devices that I was connected to before. I have only tested iOS, as I do not own an Android device.

`Get being connected device list` and `Get once connected devices' scan results` are different

You'd better file another issue if you mean the latter

Sunbreak, 2022-05-06:
> Hi Sunbreak,
> 
> I'm working on an iOS-App.
> 
> Thank you

I'd like to implement `Get being connected device list` this month

BTW. `iOS`/`macOS` could not get connected BLE devices that implement HID protocol via public APIs. Make sure before you try

Sunbreak, 2022-05-06:
A further reading: https://punchthrough.com/core-bluetooth-basics/

> In some cases, like for HID devices, once a peripheral is bonded, iOS will automatically connect to it whenever it sees the peripheral advertising. This behavior occurs independently of any app, and a peripheral can be connected to an iOS device, but not connected to the app that originally established the bond. If a bonded peripheral disconnects from an iOS device and then reconnects at the iOS level, the app will need to retrieve the peripheral object (`retrieveConnectedPeripherals(with[Services/Identifiers]:`) and explicitly connect again through a CBCentralManager to establish an app-level connection. To retrieve your device with this method, you must specify either the Apple-assigned device identifier from the previously-returned CBPeripheral object or at least one service it contains.

farhodyusupov, 2022-10-11:
> > Hi Sunbreak,
> > I'm working on an iOS-App.
> > Thank you
> 
> I'd like to implement `Get being connected device list` this month
> 
> BTW. `iOS`/`macOS` could not get connected BLE devices that implement HID protocol via public APIs. Make sure before you try

Hello Sunbreak,
did you implement this? I need solution for it. I really appreciate your response.

chan150, 2023-07-21:
I want to get a list of all devices which is already connected.

Most ble libraries have been supported it.

For example, in Flutter Blue Plus library,
```
package:flutter_blue_plus/flutter_blue_plus.dart 
Future<List<BluetoothDevice>> get connectedDevices

FlutterBluePlus.instance.connectedDevices;
```