woodemi/quick_blue #102: LateInitializationError

ftrl, 2022-05-05:
Hello,
I try to use the package but from the first line, for example :

`final b=await QuickBlue.isBluetoothAvailable();`

the error occurs:
```
E/flutter (28161): [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: LateInitializationError: Field '_instance@1480118451' has not been initialized.
E/flutter (28161): #0      QuickBluePlatform._instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart)
E/flutter (28161): #1      QuickBluePlatform.instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart:29:44)
E/flutter (28161): #2      QuickBlue.isBluetoothAvailable (package:quick_blue/quick_blue.dart:13:67)
E/flutter (28161): #3      BleController.scan (package:clickygo/pages/test.dart:331:29)
etc
```

It looks like in QuickBluePlatform, _instance is not initialized :

```
  static late QuickBluePlatform _instance;
  static QuickBluePlatform get instance => _instance;
```
Any idea ?
Thanks

Sunbreak, 2022-05-05:
Could you try `WidgetsFlutterBinding.ensureInitialize();` before `QuickBlue` APIs?

ftrl, 2022-05-05:
In the meantime, I found :

If I do :
````
flutter pub add quick_blue
````
It installs :
````
quick_blue: ^0.3.1+3
````
Changing in the file pubspec.yaml by 
````
quick_blue: ^0.4.1+1
````
it is ok

Sunbreak, 2022-05-06:
According to https://github.com/woodemi/quick_blue/issues/88, cleaning cache in `.dart_tool` may help

closed 2022-05-06 by Sunbreak