woodemi/quick_blue #108: Support advanced methods for linux

CodeDoctorDE, 2022-05-23:
Related to #29 
Currently the linux implementation only supports `isBluetoothAvailable` and `startScan/stopScan`.
It would be useful if linux supports all other methods that windows, android and ios supports.

Sunbreak, 2022-05-23:
I could add some implementation but am unable to test. Could you test on Linux?

CodeDoctorDE, 2022-05-23:
Yeah sure.
There are multiple options how to test it:
* Windows Subsystem for Linux (https://docs.microsoft.com/en-us/windows/wsl/install)
* A live cd/usb where you don't need to install (https://ubuntu.com/tutorials/try-ubuntu-before-you-install#1-getting-started)
* A vm (like virtualbox https://www.virtualbox.org/)

Sunbreak, 2022-05-24:
Yep, I see. I've a lot of devices but locked down inside my home without those testable devices

Maybe this weekend

MGlolenstine, 2022-06-15:
There is #99 waiting for further elaboration.
It does implement write and read.

Sunbreak, 2022-06-19:
Fixed by https://github.com/woodemi/quick.flutter/pull/30

closed 2022-06-19 by Sunbreak