woodemi/quick_blue #113: Messages (platform) should be checked for null

flukejones, 2022-06-01:
Version:  `quick_blue: ^0.5.0-dev.0`

```
D/BluetoothGatt(24906): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(24906): onSearchComplete() = Device=08:BE:AC:27:1A:38 Status=0
V/QuickBluePlugin(24906): onServicesDiscovered 08:BE:AC:27:1A:38 0
V/QuickBluePlugin(24906): Service 00000000-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(24906):     Characteristic 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(24906):         Descriptor 00002902-0000-1000-8000-00805f9b34fb

======== Exception caught by services library ======================================================
The following _TypeError was thrown during a platform message callback:
type 'Null' is not a subtype of type 'List<dynamic>'

When the exception was thrown, this was the stack: 
#0      MethodChannelQuickBlue._handleConnectorMessage (package:quick_blue_platform_interface/method_channel_quick_blue.dart:82:23)
```
Basic code for the service handler is:

```dart
  QuickBlue.setServiceHandler((String deviceId, String serviceId) {
    if (serviceId != BLE_SERVICE_UUID) {
      logger.error('Machine does not have BLE service: $deviceId, $serviceId');
      _peripherals.removeWhere((key, value) => value.deviceId == deviceId);
    } else if (serviceId == BLE_SERVICE_UUID) {
      for (var k in _peripherals.entries ) {
        if (k.value.deviceId == deviceId) {
          _bleConnectionEvents.add(EventMachineFound(k.key, null, true));
          return;
        }
      }
    }
    logger.info('_handleServiceDiscovery $deviceId, $serviceId');
  });
```

Sunbreak, 2022-06-01:
> ```
> #0      MethodChannelQuickBlue._handleConnectorMessage (package:quick_blue_platform_interface/method_channel_quick_blue.dart:82:23)
> ```

Is it `List<String> characteristics = (message['characteristics'] as List).cast();`?

Last time I test it could handle the usecase that service has no characteristics

Any the log above show characteristics

flukejones, 2022-06-01:
Yeah I'm not actually sure what is happening here. V0.4.x works fine but 0.5.x is getting the above error. Once I complete some more tasks for the switch to this library I'll see if I can have a closer look.

hpstuff, 2022-06-02:
@Sunbreak I actually have the same problem on windows the `quick_blue_platform_interface` is version 0.3.2 but there the code is expecting the services to be received in on batch 
```dart
List<dynamic> services = message['services'];
  for (var s in services) {
    onServiceDiscovered?.call(deviceId, s);
}
```

Sunbreak, 2022-06-02:
Released `quick_blue 0.5.0-dev.2` should solve this

closed 2022-06-02 by Sunbreak