woodemi/quick_blue #115: Android device disconnect error

flukejones, 2022-06-03:
First connection

```
V/QuickBluePlugin(30077): onScanResult: 1 + ScanResult{device=08:BE:AC:27:1A:38, scanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mServiceSolicitationUuids=[], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=RAXA20ASB09717], rssi=-80, timestampNanos=957351442689907, eventType=27, primaryPhy=1, secondaryPhy=0, advertisingSid=255, txPower=127, periodicAdvertisingInterval=0}
I/flutter (30077): DEBUG: BLE: Found: RAXA20ASB09717, 08:BE:AC:27:1A:38 [(package:jte_cloud/services/sv_machine_ble.dart:168:21)]
package:jte_cloud/services/sv_machine_ble.dart:168
D/BluetoothGatt(30077): connect() - device: 08:BE:AC:27:1A:38, auto: false
D/BluetoothGatt(30077): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
V/QuickBluePlugin(30077): onConnectionStateChange: device(08:BE:AC:27:1A:38) status(0), newState(2)
D/BluetoothGatt(30077): configureMTU() - device: 08:BE:AC:27:1A:38 mtu: 512
I/flutter (30077): DEBUG: BLE: CONNECTION: 08:BE:AC:27:1A:38, connected [(package:jte_cloud/services/sv_machine_ble.dart:96:17)]
package:jte_cloud/services/sv_machine_ble.dart:96
D/BluetoothGatt(30077): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(30077): onConfigureMTU() - Device=08:BE:AC:27:1A:38 mtu=512 status=0
D/BluetoothGatt(30077): discoverServices() - device: 08:BE:AC:27:1A:38
D/BluetoothGatt(30077): onSearchComplete() = Device=08:BE:AC:27:1A:38 Status=0
V/QuickBluePlugin(30077): onServicesDiscovered 08:BE:AC:27:1A:38 0
D/BluetoothGatt(30077): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=39 latency=0 timeout=500 status=0
I/flutter (30077): DEBUG: BLE: received: 08:BE:AC:27:1A:38, 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, þ!t12:836B5BCDA20E9C4E4C0E2A49DD79E2AD;2283ÿ [(package:jte_cloud/services/sv_machine_ble.dart:123:21)]
```

Disconnect shows two events and two failures

```
D/BluetoothGatt(30077): cancelOpen() - device: 08:BE:AC:27:1A:38
D/BluetoothGatt(30077): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
V/QuickBluePlugin(30077): onConnectionStateChange: device(08:BE:AC:27:1A:38) status(0), newState(0)
D/BluetoothGatt(30077): cancelOpen() - device: 08:BE:AC:27:1A:38
E/flutter (30077): [ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: PlatformException(IllegalArgument, Unknown deviceId: 08:BE:AC:27:1A:38, null, null)
E/flutter (30077): [ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: PlatformException(IllegalArgument, Unknown deviceId: 08:BE:AC:27:1A:38, null, null)
```

Now I have two streams on re-connect.

```
I/flutter (30077): DEBUG: BLE: CONNECTION: 08:BE:AC:27:1A:38, disconnected [(package:jte_cloud/services/sv_machine_ble.dart:96:17)]
package:jte_cloud/services/sv_machine_ble.dart:96
V/QuickBluePlugin(30077): onScanResult: 1 + ScanResult{device=08:BE:AC:27:1A:38, scanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mServiceSolicitationUuids=[], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=RAXA20ASB09717], rssi=-68, timestampNanos=957445858026912, eventType=27, primaryPhy=1, secondaryPhy=0, advertisingSid=255, txPower=127, periodicAdvertisingInterval=0}
I/flutter (30077): DEBUG: BLE: Found: RAXA20ASB09717, 08:BE:AC:27:1A:38 [(package:jte_cloud/services/sv_machine_ble.dart:168:21)]
package:jte_cloud/services/sv_machine_ble.dart:168
D/BluetoothGatt(30077): connect() - device: 08:BE:AC:27:1A:38, auto: false
D/BluetoothGatt(30077): onClientConnectionState() - status=0 clientIf=10 device=08:BE:AC:27:1A:38
V/QuickBluePlugin(30077): onConnectionStateChange: device(08:BE:AC:27:1A:38) status(0), newState(2)
D/BluetoothGatt(30077): configureMTU() - device: 08:BE:AC:27:1A:38 mtu: 512
I/flutter (30077): DEBUG: BLE: CONNECTION: 08:BE:AC:27:1A:38, connected [(package:jte_cloud/services/sv_machine_ble.dart:96:17)]
package:jte_cloud/services/sv_machine_ble.dart:96
D/BluetoothGatt(30077): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(30077): onConfigureMTU() - Device=08:BE:AC:27:1A:38 mtu=512 status=0
D/BluetoothGatt(30077): discoverServices() - device: 08:BE:AC:27:1A:38
D/BluetoothGatt(30077): onSearchComplete() = Device=08:BE:AC:27:1A:38 Status=0
V/QuickBluePlugin(30077): onServicesDiscovered 08:BE:AC:27:1A:38 0
D/BluetoothGatt(30077): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=39 latency=0 timeout=500 status=0
```

I don't know how on earth I'm getting two disconnection events. But regardless, I end up with 2 or more streams outputting to the handlers on reconnection. The same happens for if I turn the device off and reconnect.

I don't get these issues on iOS.

flukejones, 2022-06-03:
If I set:
```dart
      QuickBlue.setServiceHandler(_handleServiceDetect);
      QuickBlue.setValueHandler(_handleBleData);
```
then on second connection I get two connection events.

If I set null after first disconnect, I get single connection event (from quick blue). I'm not sure what to make of this.

Sunbreak, 2022-06-03:
> D/BluetoothGatt(30077): cancelOpen() - device: 08:BE:AC:27:1A:38
> D/BluetoothGatt(30077): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
> V/QuickBluePlugin(30077): [...]

Your device disconnected itself somehow

Android phone and BLE API stacks have a lot of problems.

You have to figure it out yourself, probably authentication or configuration(i.e. MTU negotiation) failure

Ref: https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

flukejones, 2022-06-07:
Thanks for the article. Looks like good info for another issue I have on the server device.

I solved the issue. On disconnect I had to do:
```dart
await QuickBlue.setNotifiable(
        _peripherals[_connectedMachine].deviceId, BLE_SERVICE_UUID, BLE_CHARACTERISTIC_UUID, BleInputProperty.disabled);
```

I noticed the issues were occurring approximately after the call to set:
```dart
QuickBlue.setNotifiable(
              deviceId, BLE_SERVICE_UUID, BLE_CHARACTERISTIC_UUID, BleInputProperty.notification);
```

so on a hunch I disabled on disconnect and yeah...

So I guess the final step to closing this would be:

- Is this a known thing? If not, should it be added to the readme or in a doc-comment?
- If not, perhaps this is something that can be done by `quick_blue` in the android platform part on disconnect?

Sunbreak, 2022-06-07:
I'm not sure about your point

If devices disconnected, no need to `setNotifiable(disable)`. No connection no notification

flukejones, 2022-06-07:
I mean that I have to set that *before* I disconnect the device, or quick_blue somehow ends up with duplicated incoming events.

If I force a disconnect by turning the machine off, then I can't do `setNotifiable(disable)` and end up with bad streams along with the one good stream, shown as:

```
D/BluetoothGatt(19396): connect() - device: 08:BE:AC:27:1A:38, auto: false
D/BluetoothGatt(19396): registerApp()
D/BluetoothGatt(19396): registerApp() - UUID=a3a8e3ff-2bfc-4c06-b2da-6645c9b25968
I/flutter (19396): DEBUG: Connecting to RAXA20ASB09717 via BLE [(package:jte_cloud/bloc/bl_machine_io.dart:166:21)]
D/BluetoothGatt(19396): onClientRegistered() - status=0 clientIf=10
I/flutter (19396): DEBUG: BLE scan stopped [(package:jte_cloud/services/sv_machine_ble.dart:164:19)]
D/BluetoothAdapter(19396): isLeEnabled(): ON
I/flutter (19396): stopScan invokeMethod success
D/BluetoothGatt(19396): onClientConnectionState() - status=0 clientIf=10 device=08:BE:AC:27:1A:38
V/QuickBluePlugin(19396): onConnectionStateChange: device(08:BE:AC:27:1A:38) status(0), newState(2)
I/flutter (19396): INFO: BLE: CONNECTED!!! [(package:jte_cloud/services/sv_machine_ble.dart:91:18)]
D/BluetoothGatt(19396): configureMTU() - device: 08:BE:AC:27:1A:38 mtu: 512
D/BluetoothGatt(19396): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(19396): onConfigureMTU() - Device=08:BE:AC:27:1A:38 mtu=512 status=0
D/BluetoothGatt(19396): discoverServices() - device: 08:BE:AC:27:1A:38
I/flutter (19396): INFO: BLE: MTU Okay: 512 [(package:jte_cloud/services/sv_machine_ble.dart:97:20)]
D/BluetoothGatt(19396): onSearchComplete() = Device=08:BE:AC:27:1A:38 Status=0
V/QuickBluePlugin(19396): onServicesDiscovered 08:BE:AC:27:1A:38 0
V/QuickBluePlugin(19396): Service 00000000-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(19396):     Characteristic 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(19396):         Descriptor 00002902-0000-1000-8000-00805f9b34fb
D/BluetoothGatt(19396): setCharacteristicNotification() - uuid: 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9 enable: true
I/flutter (19396): INFO: BLE: Services: 00000000-4fc7-4d40-8e54-3956e5e4ffb9, [0000afb1-4fc7-4d40-8e54-3956e5e4ffb9] [(package:jte_cloud/services/sv_machine_ble.dart:104:16)]
I/flutter (19396): INFO: BLE: device RAXA20ASB09717 [(package:jte_cloud/services/sv_machine_ble.dart:110:20)]
I/flutter (19396): INFO: BLE: device matches RAXA20ASB09717, setting notifiable [(package:jte_cloud/services/sv_machine_ble.dart:112:22)]
I/flutter (19396): INFO: BLE: done [(package:jte_cloud/services/sv_machine_ble.dart:116:22)]
D/BluetoothGatt(19396): onConnectionUpdated() - Device=08:BE:AC:27:1A:38 interval=39 latency=0 timeout=500 status=0
I/flutter (19396): DEBUG: Machine connected, begin authentication [(package:jte_cloud/bloc/bl_machine_io.dart:207:19)]
I/flutter (19396): DEBUG: Sending request: t12 null [(package:jte_cloud/bloc/bl_machine_io.dart:269:17)]
I/flutter (19396): DEBUG: Sending request via BLE [(package:jte_cloud/services/sv_machine_ble.dart:244:17)]
V/QuickBluePlugin(19396): onCharacteristicWrite 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, [-2, 63, 116, 49, 50, 59, 51, 51, 55, -1, 0, 0, 0, 0, 0] 0
D/BluetoothGatt(19396): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
V/QuickBluePlugin(19396): onConnectionStateChange: device(08:BE:AC:27:1A:38) status(0), newState(2)
V/QuickBluePlugin(19396): onCharacteristicChanged 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, [-2, 33, 116, 49, 50, 58, 68, 54, 65, 53, 54, 49, 48, 49, 50, 66, 53, 67, 65, 57, 48, 52, 67, 50, 48, 55, 52, 69, 67, 66, 48, 52, 70, 66, 69, 55, 54, 48, 59, 50, 49, 57, 57, -1]
D/BluetoothGatt(19396): configureMTU() - device: 08:BE:AC:27:1A:38 mtu: 512
D/BluetoothGatt(19396): onConfigureMTU() - Device=08:BE:AC:27:1A:38 mtu=512 status=0
I/flutter (19396): INFO: BLE: CONNECTED!!! [(package:jte_cloud/services/sv_machine_ble.dart:91:18)]
V/QuickBluePlugin(19396): onCharacteristicChanged 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, [-2, 33, 116, 49, 50, 58, 68, 54, 65, 53, 54, 49, 48, 49, 50, 66, 53, 67, 65, 57, 48, 52, 67, 50, 48, 55, 52, 69, 67, 66, 48, 52, 70, 66, 69, 55, 54, 48, 59, 50, 49, 57, 57, -1]
I/flutter (19396): DEBUG: BLE: received: 08:BE:AC:27:1A:38, 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, þ!t12:D6A561012B5CA904C2074ECB04FBE760;2199ÿ [(package:jte_cloud/services/sv_machine_ble.dart:131:21)]
I/flutter (19396): DEBUG: Recieved IV: D6A561012B5CA904C2074ECB04FBE760, requesting auth [(package:jte_cloud/bloc/bl_machine_io.dart:468:21)]
I/flutter (19396): INFO: BLE: MTU Okay: 512 [(package:jte_cloud/services/sv_machine_ble.dart:97:20)]
D/BluetoothGatt(19396): discoverServices() - device: 08:BE:AC:27:1A:38
D/BluetoothGatt(19396): onSearchComplete() = Device=08:BE:AC:27:1A:38 Status=0
V/QuickBluePlugin(19396): onServicesDiscovered 08:BE:AC:27:1A:38 0
V/QuickBluePlugin(19396): Service 00000000-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(19396):     Characteristic 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9
V/QuickBluePlugin(19396):         Descriptor 00002902-0000-1000-8000-00805f9b34fb
I/flutter (19396): DEBUG: BLE: received: 08:BE:AC:27:1A:38, 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9, þ!t12:D6A561012B5CA904C2074ECB04FBE760;2199ÿ [(package:jte_cloud/services/sv_machine_ble.dart:131:21)]
I/flutter (19396): DEBUG: Recieved IV: D6A561012B5CA904C2074ECB04FBE760, requesting auth [(package:jte_cloud/bloc/bl_machine_io.dart:468:21)]
[... cut ...]
In the log above you can see I get duplicated `onClientConnectionState`, leading to the callback doing an MTU request, then the chain of calls from that. For every force disconnect I end up with an extra `onClientConnectionState` on next re-connection, and then I can't complete the internal auth (it's not BLE auth) because I end up with multiple request/response chains out of order.

By force disconnect I mean I power-off the machine. And this really is not an issue with our BLE devices.

I'm sorry I can't do much to debug this as I don't know kotlin. But it really does seem as if a stream or channel isn't closing, or being duplicated.

flukejones, 2022-06-07:
Is it at all possible that the [`remoteDevice.connectGatt`](https://github.com/woodemi/quick_blue/blob/f451df7ce21eb77074c6fdcbe2999ae7a933bbb7/quick_blue/android/src/main/kotlin/com/example/quick_blue/QuickBluePlugin.kt#L84) could be getting... I don't, held when it should be if the connection state changes under some circumstances?

Sunbreak, 2022-06-07:
> Is it at all possible that the [...]

```kt
        val gatt = if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
          remoteDevice.connectGatt(context, false, gattCallback, BluetoothDevice.TRANSPORT_LE)
        } else {
          remoteDevice.connectGatt(context, false, gattCallback)
        }
        knownGatts.add(gatt)
```

`gatt` is the return value of either `connectGatt(BluetoothDevice.TRANSPORT_LE)` or `connectGatt()`

And `knownGatts` holds that `gatt`

Sunbreak, 2022-06-07:
> D/BluetoothGatt(19396): onClientConnectionState() - status=0 clientIf=10 device=08:BE:AC:27:1A:38
> ...
> D/BluetoothGatt(19396): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38

According to https://stackoverflow.com/questions/34158315/clientif-field-in-android-bluetoothgatt-class, `clientIf` is the `scannerId`. Is it possible they belongs to two ScanResult's ?

flukejones, 2022-06-07:
> According to https://stackoverflow.com/questions/34158315/clientif-field-in-android-bluetoothgatt-class, `clientIf` is the `scannerId`. Is it possible they belongs to two ScanResult's ?

Not in my code at least. I verified making sure that only one scan was ever run, and purged the result list after scans. Same result as before, if I yank machine power I get a disconnect then on reconnect there's that same `onClientConnectionState` double-up you saw - and in fact there is an incremental numbered one added after each machine-power-off disconnect event, so this looks like the clue.

Forcing myself to learn some kotlin through this... I'm baffled by the issue. It actually looks to still be something to do with `setNotifiable`.

knownGatts is a list right? Forcing myself to learn some kotlin as I go along here... My working theory is maybe that's related or is the issue...

1. Unexpected disconnect
2. `knownGatts.remove(gatt)` is in `cleanConnection` and requires `gatt` as arg
3. `cleanConnection` is called on method `disconnect` call, and `gattCallback` on connection state change
4. Since a list can hold many of similar, could that be the issue? Not being cleaned up?

That doesn't seem likely from what I've read though.. How would I end up with previous scan results in the stream but only if it's an unexpected disconnect?

Looking at `BasicMessageChannel`, this states it is FIFO... On first connection I get:
```
D/BluetoothGatt(25446): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
```
then on third connection after (two) unexpected disconnect I get this order
```
D/BluetoothGatt(25446): onClientConnectionState() - status=0 clientIf=11 device=08:BE:AC:27:1A:38
D/BluetoothGatt(25446): onClientConnectionState() - status=0 clientIf=10 device=08:BE:AC:27:1A:38
D/BluetoothGatt(25446): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
```
somehow the previous result is being held and sent... out of order?

Setting `QuickBlue.setConnectionHandler(null) on disconnect in app then back to handler on connect action also doesn't help.

In app, stopscan cancels the scan result listener.. Stopping or starting has no effect on the above issue.

I'm almost ready to believe it's related to `setNotifiable` as that setting carries over between connections as long as the app is live, and that I had to disable on manual connect to prevent the first similar (same?) issue. I'm willing to put in the legwork but I do need some pointers since I'm neither kotlin or BLE proficient to your level.

Sunbreak, 2022-06-07:
> A side note: flutter_reactive_ble sets the notification to off/disabled on disconnect automatically:
> 
> ```
> D/BluetoothGatt(28888): setCharacteristicNotification() - uuid: 0000afb1-4fc7-4d40-8e54-3956e5e4ffb9 enable: false
> ```

A little curious how it is done. I've not idea how to disable notification after disconnected

flukejones, 2022-06-11:
I don't know sorry.

This seems to narrow down to:

Connect:
```
D/BluetoothGatt(32231): onClientConnectionState() - status=0 clientIf=5 device=08:BE:AC:27:1A:38
```
Unexpected disconnect:
```
D/BluetoothGatt(32231): onClientConnectionState() - status=8 clientIf=5 device=08:BE:AC:27:1A:38
```
Reconnect:
```
D/BluetoothGatt(32231): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
D/BluetoothGatt(32231): onClientConnectionState() - status=0 clientIf=5 device=08:BE:AC:27:1A:38
```

I see that the last line is the state from first connection.

The reconnect happens after a re-scan... The re-scan I have removes previous scan results if they are not found again. So when the unexpected disconnect happens, the previous (`clientIf=5`) result is removed from the map (the map is k=serial/name, v=result).

Update: I removed the bit where previous result is removed. There is no difference.

I've rewritten a huge chunk of code in a few different ways to narrow this down as much as I can, and it looks like `quick_blue` retains the previous results?

Oh and curiously:
```
D/BluetoothGatt( 4362): onClientConnectionState() - status=0 clientIf=9 device=08:BE:AC:27:1A:38
D/BluetoothGatt( 4362): onClientConnectionState() - status=22 clientIf=5 device=08:BE:AC:27:1A:38
```

It happens even if the scan is left running.

---

I've gone the nuclear option and on disconnect completely destroy and replace the service part of my code that deals with BLE, still no change. I can't replicate this issue using the example scan app either. I'm at a loss.

TomixUG, 2022-07-19:
Hi, I have the same issue with the duplicated handleValueChange events after enabling a notification.

GorIvanov, 2023-06-09:
On Disconnect (Android) there is error

```
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: PlatformException(IllegalArgument, Unknown deviceId: 18:04:ED:06:BD:C6, null, null)
E/flutter (25794): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:607:7)
E/flutter (25794): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:167:18)
E/flutter (25794): <asynchronous suspension>
E/flutter (25794): 

```