woodemi/quick_blue #117: Unhandled Exception: LateInitializationError: Field '_instance@609118451' has not been initialized

RRohitM, 2022-06-10:
Hello I have to integrate the Quick plugin for Bluetooth connection but got some error.

[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: LateInitializationError: Field '_instance@609118451' has not been initialized.
E/flutter (15409): #0      QuickBluePlatform._instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart)
E/flutter (15409): #1      QuickBluePlatform.instance (package:quick_blue_platform_interface/quick_blue_platform_interface.dart:29:44)
E/flutter (15409): #2      QuickBlue.scanResultStream (package:quick_blue/quick_blue.dart:20:30)
E/flutter (15409): #3      BluetoothPairController.getBattryLevel (package:sound_gallary/app/modules/main/bluetooth_pair/controllers/bluetooth_pair_controller.dart:176:15)
E/flutter (15409): #4      BluetoothPairView.buildWidget.<anonymous closure>.<anonymous closure> (package:sound_gallary/app/modules/main/bluetooth_pair/views/bluetooth_pair_view.dart:82:34)
E/flutter (15409): #5      CustomButton.primaryButton.<anonymous closure> (package:sound_gallary/master/generic_class/custom_button.dart:36:47)
E/flutter (15409): #6      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:198:24)
E/flutter (15409): #7      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:608:11)
E/flutter (15409): #8      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:296:5)
E/flutter (15409): #9      BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:230:7)
E/flutter (15409): #10     PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:563:9)
E/flutter (15409): #11     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:94:12)
E/flutter (15409): #12     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:139:9)
E/flutter (15409): #13     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:539:8)
E/flutter (15409): #14     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:137:18)
E/flutter (15409): #15     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:123:7)
E/flutter (15409): #16     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:439:19)
E/flutter (15409): #17     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:419:22)
E/flutter (15409): #18     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:322:11)
E/flutter (15409): #19     GestureBinding._handlePointerEventImmediately (package:flutter/src/gestures/binding.dart:374:7)
E/flutter (15409): #20     GestureBinding.handlePointerEvent (package:flutter/src/gestures/binding.dart:338:5)
E/flutter (15409): #21     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:296:7)
E/flutter (15409): #22     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:279:7)
E/flutter (15409): #23     _rootRunUnary (dart:async/zone.dart:1442:13)
E/flutter (15409): #24     _CustomZone.runUnary (dart:async/zone.dart:1335:19)
E/flutter (15409): #25     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1244:7)
E/flutter (15409): #26     _invoke1 (dart:ui/hooks.dart:170:10)
E/flutter (15409): #27     PlatformDispatcher._dispatchPointerDataPacket (dart:ui/platform_dispatcher.dart:331:7)
E/flutter (15409): #28     _dispatchPointerDataPacket (dart:ui/hooks.dart:94:31)

Sunbreak, 2022-06-10:
Could you post `flutter doctor -v` and version of `quick_blue`?

RRohitM, 2022-06-13:
Hello @Sunbreak please check our flutter doctor -v and quick_blue version

quick_blue: ^0.3.1+3

 Flutter (Channel stable, 2.10.4, on macOS 11.5.2 20G95 darwin-x64, locale en-GB)
    • Flutter version 2.10.4 at /Users/apple/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision c860cba910 (3 months ago), 2022-03-25 00:23:12 -0500
    • Engine revision 57d3bac3dd
    • Dart version 2.16.2
    • DevTools version 2.9.2

[!] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at /Users/apple/Library/Android/sdk
    ✗ cmdline-tools component is missing
      Run `path/to/sdkmanager --install "cmdline-tools;latest"`
      See https://developer.android.com/studio/command-line for more details.
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/macos#android-setup for more details.

[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.3

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)

[✓] VS Code (version 1.64.2)
    • VS Code at /Users/apple/Downloads/Visual Studio Code.app/Contents
    • Flutter extension can be installed from:
      🔨 https://marketplace.visualstudio.com/items?itemName=Dart-Code.flutter

[✓] Connected device (2 available)
    • SM A515F (mobile) • RZ8R21Z8TYT • android-arm64  • Android 11 (API 30)
    • Chrome (web)      • chrome      • web-javascript • Google Chrome 102.0.5005.115
    ! Error: iPhone is not connected. Xcode will continue when iPhone is connected. (code -13)

[✓] HTTP Host Availability
    • All required HTTP hosts are available

! Doctor found issues in 1 category.

Sunbreak, 2022-06-13:
Related to

https://github.com/woodemi/quick_blue/blob/1083edc15b0076a929969987fe3a72b5ce71350b/quick_blue/lib/quick_blue.dart#L17-L36

It is changed in new repo https://github.com/woodemi/quick.flutter/tree/master/packages/quick_blue

Could you try that locally?

RRohitM, 2022-06-14:
Thanks, @Sunbreak it's working now but I have one more issue faced when I click on connect button and try to connect the device got a disconnected state in handleConnectionChange and the device is not connected what can I do?

Sunbreak, 2022-06-14:
You'd better open another issue or ask on discord

https://discord.gg/zFUK9Tfk

closed 2022-06-14 by Sunbreak