woodemi/quick_blue #119: Microsoft.Windows.CppWinRT.props was not found in Windows 10

dirablue, 2022-06-19:
Hi I testeed this plugin but error occurred on Windows 10.

Do you have any idea to fix this?

quick_blue: ^0.5.0-dev.2
windows 10
Visual Studio Community 2022 ( I also installed 2019 )

this error message was displayed when I started bulding flutter project.

```
Launching lib\main.dart on Windows in debug mode...
Building Windows application...
<path of project>\build\windows\plugins\quick_blue_windows\quick_blue_windows_plugin.vcxproj(258,3): error MSB4019: The imported project "<path of project>\build\windows\packages\Microsoft.Windows.CppWinRT\build\native\Microsoft.Windows.CppWinRT.props" was not found. Confirm that the expression in the Import declaration "<path of project>\build\windows\packages\Microsoft.Windows.CppWinRT\build\native\Microsoft.Windows.CppWinRT.props" is correct, and that the file exists on disk.
Exception: Build process failed.
```

It seems the code tried to find "build/windows/packages" folder, but there is no packages folder in my project.

[image]

Sunbreak, 2022-06-20:
https://github.com/woodemi/quick_notify/discussions/8

Sunbreak, 2022-06-21:
Could you try `local_auth`?

The `CMakeLists.txt` is inspired by https://github.com/flutter/flutter/issues/32719#issuecomment-1136341840

dirablue, 2022-06-21:
@Sunbreak 

Thank you for sharing this.

I could installed CppWinRT with these stesps following.

1. set Package Sources of nuget in Visual Studio 2022

[image]

2. Add to last line of CMakeLists.txt in <workspace>/windows folder

```
set(CPPWINRT_VERSION "2.0.220608.4")

find_program(NUGET nuget)
if (NOT NUGET)
    message("Nuget.exe not found, trying to download or use cached version.")
endif()

execute_process(COMMAND
    ${NUGET} install Microsoft.Windows.CppWinRT -Version ${CPPWINRT_VERSION} -OutputDirectory packages
    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
    RESULT_VARIABLE ret)
if (NOT ret EQUAL 0)
    message(FATAL_ERROR "Failed to install nuget package Microsoft.Windows.CppWinRT.${CPPWINRT_VERSION}")
endif()

```

but after the app starts, another error occurred and crashed if I used this plugin in flutter.
when I don't use this plugin, the app starts correctlly.

```
[   +7 ms] Error connecting to the service protocol: failed to connect to
           http://127.0.0.1:63658/W0h2o9fuOSI=/
```

should I prepare something if I use this plugin ..?

dirablue, 2022-06-21:
https://github.com/woodemi/quick_blue/issues/80#issuecomment-1086327365

it seems there is a same thing.
let me check this..

dirablue, 2022-06-21:
It works after I switched from Visual Studio 2022 to 2019.
Thanks !

closed 2022-06-21 by Sunbreak