woodemi/quick_blue #120: Possibility to update to bluez 0.8.0 package

h3x4d3c1m4l, 2022-07-03:
As `quick_blue_linux` currently uses `bluez` 0.7.9 (which in turn uses outdated `dbus`, which in turn uses outdated `xml`), it blocks the usage of recent versions of libraries like `flutter_svg` and `flutter_native_splash`.

Would it be possible to upgrade `bluez` to 0.8.0?

Sunbreak, 2022-07-04:
Try https://github.com/woodemi/quick.flutter/tree/master/packages/quick_blue

h3x4d3c1m4l, 2022-07-06:
Hi @Sunbreak I checked both `master` and `develop` branches but both still use the 0.7.9 version of `bluez`

Sunbreak, 2022-07-07:
You could try `dependency_overrides` in your APP's `pubspec.yaml`

```
dependencies:
  quick_blue: any

dependency_overrides:
  bluez: 0.8.0
```