woodemi/quick_blue #124: location permission is always denied.

ext-danny, 2022-07-30:


Sunbreak, 2022-07-31:
You dont need location permission on Android 12+

ext-danny, 2022-08-01:
Thanks but I would like to also add that when I run the quick_blue example I can't connect to android 12+ and this is the error I get.
`
Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10146, packageName = com.example.quick_blue_example, attributionTag = null, token = android.os.BinderProxy@39e12e, next = null }: GattService registerScanner`

Sunbreak, 2022-08-04:
`BLUETOOTH_SCAN` has nothing to do with location on Android 12+

---

I'm not sure what is `connect to android 12+`

Do you run `quick_blue` example as central on Android 12+ device, or run example on another device and trying to connect a perpheral on Android 12+ device?

ext-danny, 2022-08-04:
My apologies, I didn't put that right, so basically I tried running the existing quick_blue example on an android 12 device as a central and try to connect to available peripherals. My doubts are

1. Do we need to add any permission or modify the code or will the example work right out of the box?
2. What I was expecting from the example is that when I tap on scan I will be able to see list of available Peripherals which I can connect?

Sunbreak, 2022-08-05:
I'm busy on https://github.com/woodemi/quick.flutter

I'll leave it open until someone find the answer