woodemi/quick_blue #125: Bad deviceId causes hard crash

Erhannis, 2022-08-17:
Windows 10, quick_blue 0.4.1+1, flutter 3.3.0-0.2.pre

When I try to `connect` to a `deviceId` that isn't valid, on Windows at least, (such as "asdf") my app crashes, with an error like:
```
Microsoft Visual C++
---
Debug Error!

Program:
...ts\flutter_tests\build\windows\runner\Debug\flutter_tests.exe

abort() has been called

(Press Retry to debug the application)

[Abort] [Retry] [Ignore]
```

If I wait, the app crashes and Flutter says it lost connection.  If I click a button, same thing.

(Note that if I give a device id that is valid but doesn't match a device, it just gives me a normal error, rather than crashing.)

Sunbreak, 2022-08-17:
Try https://github.com/woodemi/quick.flutter

Erhannis, 2022-08-18:
Well, it didn't give me a popup, this time; it just made a alert noise and crashed.

ext-danny, 2022-08-24:
@Sunbreak Hi, quick question.
will https://github.com/woodemi/quick.flutter this be published as a completely new plugin so that we can add it in the pubspec and use it OR will you be updating the existing quick_blue plugin.