woodemi/quick_blue #127: The Flutter application cannot start if it has this plugin and there is no bluetooth dongle inserted into the Windows computer.

alevlako, 2022-08-22:
How to reproduce:
1. Run flutter create testapp
2. flutter pub add quick_blue, you don't need any usage of plugin in your code, this is not a point
3. Run flutter app with bluetooth dongle (Ok)
4. Remove bluetooth dongle from computer with started app (Ok)
5. Close flutter app
6. Run flutter app without bluetooth dongle (App cannot start)

Bikram40, 2022-11-10:
i am facing the same issue can not run build windows application with this plugin any solution ?

alevlako, 2022-11-10:
As far as I understand, the application crashes on a level inaccessible for flutter when it try to load this plugin in the absence of a bluetooth device in OS. I suggest, the problem can only be solved inside the plugin by adding a check for the presence of a BT device before any attempt calling it.

RobinHeim, 2023-02-22:
I faced the same issue and did a little debugging.
I figured out, that the InitializeAsync() function  (which is called wheter the plugin is used or not) in the quick_blue_windows_plugin.cpp causes the crash. It calls bluetoothAdapter.GetRadioAsync(), but bluetoothAdapter is null when there is no bluetooth adapter available in the system. You can also easily simulate this by disabling the bluetooth adapter in the device manager.

I could fix it by simply adding a null check:
Old:
`winrt::fire_and_forget QuickBlueWindowsPlugin::InitializeAsync() {
  auto bluetoothAdapter = co_await BluetoothAdapter::GetDefaultAsync();
  bluetoothRadio = co_await bluetoothAdapter.GetRadioAsync();
}`

New:
`winrt::fire_and_forget QuickBlueWindowsPlugin::InitializeAsync() {
  auto bluetoothAdapter = co_await BluetoothAdapter::GetDefaultAsync();
  if (bluetoothAdapter) {
    bluetoothRadio = co_await bluetoothAdapter.GetRadioAsync();
  }
}`

However, take this advise with a grain of salt as I am by no means a C++ or Windows System programmer. There might be a better solution, but I got it working with this fix.

alevlako, 2023-02-25:
> I faced the same issue and did a little debugging. I figured out, that the InitializeAsync() function (which is called wheter the plugin is used or not) in the quick_blue_windows_plugin.cpp causes [...]

Great, it workы!
I forked master branch and add your fix to code.
Now anyone suffering from this problem can add a plugin to their project's pubspec.yaml file like this:

`quick_blue:
    git:
      url: https://github.com/alevlako/quick_blue.git
      ref: master #branch name
      path: quick_blue #Folder Path on Github`

gtedone, 2023-10-20:
Thanks for this fix, it works like a charm!

> However, take this advise with a grain of salt as I am by no means a C++ or Windows System programmer. There might be a better solution, but I got it working with this fix.

In my opinion it is a simple and effective solution: if nothing else, with this patch you can launch the application even if bluetooth is disabled. Then you can check the controller activation after the UI comes up (maybe via [isBluetoothAvailable](https://pub.dev/documentation/quick_blue/latest/quick_blue/QuickBlue/isBluetoothAvailable.html) static method.

> Now anyone suffering from this problem can add a plugin to their project's pubspec.yaml file like this:

@alevlako, you can get it working only if you apply your fork as dep override, rather than the entire package replacement:

```yaml
dependencies:
  flutter:
    sdk: flutter
  quick_blue: ^0.5.0-dev.2 # leave here the federated plugin

...

dependency_overrides: 
  quick_blue_windows:
    git:
      url: https://github.com/alevlako/quick_blue.git
      ref: issue_127_fix # branch name
      path: quick_blue_windows # Folder Path on Github
```

Otherwise flutter cannot build the expected patch, since the resulting dep chain is not the desired one.

In other words, I think your pull-request is to be made for the [quick_blue_windows](https://pub.dev/packages/quick_blue_windows/versions/0.4.0-dev.0) sub-project... :wink: