woodemi/quick_blue #139: Fatal error: Unexpectedly found nil while unwrapping an Optional value

kiranigeektech, 2023-01-03:
Hi

I'm trying to use writeValue() function to write binary code and i'm using following code.

     var value = Uint8List.fromList(hex.decode(binaryCode.text));
     QuickBlue.writeValue(
      widget.deviceId, 'e5f5f000-c0ec-46ba-bd1f-b4c9f0da4208', 'e5f50001-c0ec-46ba-bd1f-b4c9f0da4208',
      value, BleOutputProperty.withResponse);

But i'm getting following error.

quick_blue_macos/QuickBlueMacosPlugin.swift:32: Fatal error: Unexpectedly found nil while unwrapping an Optional value

Can you please help me to sort out this issue.

Thank you in advance.

Bhaveshsen29, 2024-02-16:
We are also facing same error? any resolution for this

JozehanGrobler, 2024-04-03:
Did support for this repo die,
Deep into an inherited project, where I get the same error as above
Anyone have any luck with getting arround this
Or have a good suggestion on how to replace  Quick_Blue

LucaTrip, 2024-04-29:
I've the same problem with the version **quick_blue: 0.5.0-dev.2**. Some ideas?