woodemi/quick_blue #143: [Android]PlatformException(Characteristic unavailable, null, null, null)

ankita-kanchan, 2023-04-25:
quick blue irregularly throws **PlatformException(Characteristic unavailable, null, null, null)**

E/flutter (21750): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(Characteristic una
vailable, null, null, null)
E/flutter (21750): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653)
E/flutter (21750): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:296)
E/flutter (21750): <asynchronous suspension>

GorIvanov, 2023-06-08:
I have found solution when debuging.
First we should write descyptor.
```
QuickBlue.writeValue(
        selectedDeviceAddress!.address!,
        uuidEEG,
        uuidEEGCtrl,
        Uint8List.fromList([0x01, 0x01]),
        BleOutputProperty.withResponse,
      ).
```

Next we call setNotifiable **but** there should be a timeout... I take 1 sec.
So this works for me:

```
      QuickBlue.writeValue(
        selectedDeviceAddress!.address!,
        uuid1,
        uuid2,
        Uint8List.fromList([0x01, 0x01]),
        BleOutputProperty.withResponse,
      ).then((value) => {
            Timer(
                Duration(milliseconds: 1000),
                () => {
                      QuickBlue.setNotifiable(
                        selectedDeviceAddress!.address!,
                              uuid1,
                                uuid3,
                        BleInputProperty.notification,
                      ).then((value) => {}),
                    }),
          });
```

SahilSharma2710, 2023-06-19:
@ankita-kanchan hey , Ankita can u provide a running example for this package in android . It's urgent. I am unable to run this package on android

ethmth, 2023-06-20:
Seconded on the example! I can't seem to get this to work on Android either.

SahilSharma2710, 2023-06-20:
@ethmth  i am able to run this package on android now
This  following repo was helpful
  [https://github.com/MatheusJulidori/dart-flutter/tree/main/bluetooth_app](url)

SahilSharma2710, 2023-06-21:
@ethmthcan you update me once if u got any solution of ur problem , i am having same issues.