woodemi/quick_blue #144: Does it support connecting multiple devices ？

qq1023308356, 2023-05-06:
Does it support connecting multiple devices ？

MrCsabaToth, 2023-06-14:
Inspecting the plugin as a Flutter Blue Plus user, I wonder the same. Based on the fact that the `setNotifiable` API call is global / static, I have a feeling that right now no. Also see https://github.com/woodemi/quick_blue/issues/140

SahilSharma2710, 2023-06-19:
hey @qq1023308356 @MrCsabaToth guys can u provide any working example of this package

MrCsabaToth, 2023-06-19:
> hey @qq1023308356 @MrCsabaToth guys can u provide any working example of this package

I cannot. I'm a flutter_blue_plus user right now and will remain as such if QuickBlue cannot handle multiple devices or multiple characteristics. So far I don't have a definitive answer, just a hunch based on the plugin's API.
BTW there are examples in the plugin's pub page and in the README.

chipweinberger, 2023-07-20:
why do you think that? all of the API's take a `deviceId` parameter.