woodemi/quick_blue #150: Simple help

MosheMendel18, 2023-06-22:
I could use some help just doing a simple read and write. I'm not sure how to get the `serviceId` `characteristicId`, `BleInputProperty`, or `BleOutputProperty` to run the code in the docs. Here's what I tried. It runs without error and manages to connect, but I don't know how to see any of the data from any devices. I think the beginning and end are fine, feel free to skip to the area between the asterisks.

```
void _scanBLE() async
  {
      var scanResult = await Permission.bluetoothScan.request();
      var connRes = await Permission.bluetoothConnect.request();
      if (scanResult == PermissionStatus.granted && connRes == PermissionStatus.granted) {
        setState(() {
          QuickBlue.startScan();
          QuickBlue.scanResultStream.listen((scanResult) {
            print('onScanResult ok ${scanResult.name}');
            QuickBlue.connect(scanResult.deviceId);
            void _handleConnectionChange(String deviceId, BlueConnectionState state) {
              print('_handleConnectionChange $deviceId, $state');
            }
            QuickBlue.setConnectionHandler(_handleConnectionChange);
//**********************************************************************************************************
//**********************************************************************************************************
//**********************************************************************************************************
// In order to get the values for serviceId and characteristicId, I just tried nesting value change, write, and read. I'm sure it's wrong.
            void _handleServiceDiscovery(String deviceId, String serviceId) { 
              print('_handleServiceDiscovery $deviceId, $serviceId');
                    void _handleValueChange(String deviceId, String characteristicId, Uint8List value) {
                      print('_handleValueChange $deviceId, $characteristicId, ${hex.encode(value)}');
                      QuickBlue.setNotifiable(scanResult.deviceId, serviceId, characteristicId, true as BleInputProperty);
                      List<int> list = 'Hello'.codeUnits; // random value for Uint8List, obviously can't be expected to set the value here
                      Uint8List bytes = Uint8List.fromList(list); 
                      QuickBlue.writeValue(deviceId, serviceId, characteristicId, bytes, true as BleOutputProperty); // just casted this to get it to work, but I'm sure it's wrong
                      QuickBlue.readValue(deviceId, serviceId, characteristicId);
                      QuickBlue.setNotifiable(deviceId, serviceId, characteristicId, true as BleInputProperty); // just casted this to get it to work, but I'm sure it's wrong
                    }
              QuickBlue.setValueHandler(_handleValueChange);
            }
            QuickBlue.setServiceHandler(_handleServiceDiscovery);
            QuickBlue.discoverServices(scanResult.deviceId);
            sleep(Duration(seconds: 3));
            QuickBlue.disconnect(scanResult.deviceId);
          });
//**********************************************************************************************************
//**********************************************************************************************************
//**********************************************************************************************************
          sleep(Duration(seconds: 4)); 
        QuickBlue.stopScan();
        });
      } else {
        if(scanResult == PermissionStatus.permanentlyDenied || connRes == PermissionStatus.permanentlyDenied) {
        openAppSettings();
      }
      }
  }
``` 

I expect the main issues are between the asterisk lines. I've tried to query things, but it seems like all the methods are `void`, and setting global variables didn't seem to work. Any help would be appreciated.

SahilSharma2710, 2023-06-22:
Hey @MosheMendel18 , when we call `QuickBlue.discoverServices(deviceId);` it give us the list of all the services and their characteristics list of a particular device . You can use these `serviceId` `characteristicId` to use the any read or write operations

MosheMendel18, 2023-06-22:
> Hey @MosheMendel18 , when we call `QuickBlue.discoverServices(deviceId);` it give us the list of all the services and their characteristics list of a particular device . You can use these `serviceId` `characteristicId` to use the any read or write operations

Thanks for the reply, @SahilSharma2710.  Would you be able to explain this a little more for me? Where do those values get saved so that I can call them in another command? (From what I can tell, it's a void method.) 

For example, if I do this: 

```
            QuickBlue.discoverServices(scanResult.deviceId);
            QuickBlue.readValue(scanResult.deviceId,serviceId,characteristicId); 
```
The editor shows errors under `serviceId` and `characteristicId`, so I can't run the program.