woodemi/quick_blue #151: Newbie needs help

genee19, 2023-06-22:
Hi. I am trying to create a crossplatform app that operates with a specific Bluetooth LE device. This is my first time working with Bluetooth LE anywhere and I am having some trouble.

I have so far tried the very primitive example of scanning for devices and displaying them in a list, almost literally what you can get from the readme of `quick_blue`:

```dart
_report_line("Manual scan started");
    QuickBlue.startScan();
    QuickBlue.scanResultStream.listen((result) {
      _report_line("Manual scan result: $result");
    });
    Future.delayed(const Duration(seconds: 15), () {
      QuickBlue.stopScan();
      _report_line("Manual scan stopped");
    });
```

 I've built it for macOS (my working laptop) and Android (target platform) and nowhere it is able to see any devices. It doesn't produce any errors or permission requests, just silently waits for timeout and stops scanning. I've tried to scan for a device that I know is working and is visible to that same phone (in Settings > Bluetooth), but it doesn't show up in the list of my app.

 Do I have to request permissions myself? Maybe I miss some specific permissions in manifest? Please help me out.

SahilSharma2710, 2023-06-22:
hey @genee19 , try this quick_blue_example , and try to debug app on real device
 [https://github.com/SahilSharma2710/quick_blue_example_2.0](url)

genee19, 2023-06-22:
Hey, thanks for the answer. Worth mentioning that I test this on a real Android phone, a Galaxy A22, and I upload a built APK onto it via Bluetooth file transfer, so obviously BT is alive and kicking on both Mac and the phone.

I tried your example and it works like magic. I am trying to understand just which part I am missing.

SahilSharma2710, 2023-06-22:
I am also new to this thing , I also have couple of questions , have you tried `flutter_blue_plus` package ? And do u have proper understanding of how things went in this BLE?