woodemi/quick_blue #163: Repository Still Active?

bdebyl, 2023-11-08:
There hasn't been any interaction with the maintainers of the repository with issues, or any other contributions including commits on any branch.

It seems as though this repository is no longer maintained and I would like to start a discussion surrounding this.

Personally, I have found a fork from Pison to be the most active at this point in time: https://github.com/pisontechnology/quick_blue

If I am wrong and this repository is planned to continue to be maintained I'd love to know!

LukeFernandes, 2024-01-16:
Also interested in whether this repository is no longer maintained (for good) and which fork is best,

Unfortunately, the Pison fork you've linked to returns errors when I try to use it, beginning with the following:
```
ERROR: ../../qblue/quick_blue/quick_blue/lib/quick_blue.dart:50:5: Error: Type 'ScanFilter' not found.
ERROR:     ScanFilter scanFilter = const ScanFilter(),
ERROR:     ^^^^^^^^^^
ERROR: ../../qblue/quick_blue/quick_blue/lib/quick_blue.dart:114:17: Error: Type 'BleL2capSocket' not found.
ERROR:   static Future<BleL2capSocket> openL2cap(String deviceId, int psm) =>
ERROR:                 ^^^^^^^^^^^^^^
ERROR: ../../qblue/quick_blue/quick_blue/lib/quick_blue.dart:50:5: Error: 'ScanFilter' isn't a type.
ERROR:     ScanFilter scanFilter = const ScanFilter(),
ERROR:     ^^^^^^^^^^
ERROR: ../../qblue/quick_blue/quick_blue/lib/quick_blue.dart:50:35: Error: Couldn't find constructor 'ScanFilter'.
ERROR:     ScanFilter scanFilter = const ScanFilter(),
ERROR:                                   ^^^^^^^^^^
```
The original repo doesn't give these errors, nor does the more up-to-date [MinkTec](https://github.com/MinkTec/quick_blue) one.

talpadk, 2024-05-10:
I'm also interested in a working cross platform ble implementation (with working Linux and Android support), preferably with more than one maintainer.

I haven't tried out the MinkTec repo, but at least it seems to have code for more than getting scan results under Linux